WEC-Sim/WEC-Sim #3: block diagram parameter error

brauliobarahona, 2015-04-23:
Hi guys,

I am trying to simply run the model in `RM3.slx` as it is setup in the script `wecSimInputFile.m`, both of them located in the `applications` folder.

I am using Matlab R2015a, with a trial version of Simulink. I get the following error:

```
Error using simulationClass/loadSimMechModel (line 101)
block_diagram does not have a parameter named 'SimMechanicsOpenEditorOnUpdate'

Error in wecSim_RunHere (line 93)
simu.loadSimMechModel(simu.simMechanicsFile);
```

lawsonro3, 2015-04-23:
Are you sure that you have installed SimMechanics in 2015a - we just tested it and it is working on our computers.

Can you type `ver` in the command window to make sure you have SimScape and SimMechanics installed?

brauliobarahona, 2015-04-23:
I have a trial version of Simulink, with many libraries, but not SimScape nor SimMechanics.Those seem to be more *fancy* libraries.

So, I guess it will not work only with Simulink. Which libraries do I need, only SimScape and SimMechanics?

closed 2015-04-23 by lawsonro3