WEC-Sim/WEC-Sim #5: Random seed in batch simulations

brauliobarahona, 2015-07-04:
Hey guys,

I run in the problem that using `waves.randPreDefined = 0;` sets the random generator seed to 0, which effectively defines a **deterministic** sequence of numbers drawn from the default random number generator algorithm. This deterministic sequence will repeat itself in any instance of Matlab, when re-started, or when the random number generator is called again with the same seed number ( `>> rng(0)` )

I think the description in the manual and the annotations in `waveClass.m` are bit misleading:

> Note that, by default, the random wave phase for irregular waves are generated arbitrarily (waves.randPreDefined=0). If the user species waves.randPreDefined=1 in the input file, the random phase of the waves will be generated using the rand function in MATLAB with a seed value of 1. This gives the user an option to generate the same random wave every time if needed.

As far as I can see, either of the options (`waves.randPreDefined = 0;` or `waves.randPreDefined = 1;`) will give a deterministic sequence of numbers from the random number generator. One with a seed number equal to 0 (because it uses directly `rand`, which will use `rng` with default values) and the other equal to 1. Every time you call the random number generator `>> rng( seed_number )` it will start the same deterministic sequence corresponding to the given `seed_number`.

Ok, this is getting too long, but I hope it is clear... So, when `rand` is called for the first time it starts the random number generator, succesive calls to `rand` will contine drawing numbers from the sequence defined by the seed, until `rng` is called again to reset the seed number.

The problem then is that running batch simulations *on different instances of Matlab* with `waves.randPreDefined = 0;` will output simulations on each instance with the same sequence of random phase. Because the first simulation of each batch, in each Matlab instance, starts `rng` with the same seed.

The solution is to give a different `seed_number` (i.e., `rng(seed_number)` or `waves.randPreDefined = seed_number;`) for each batch. The easiest way to do this is to use `rng('shuffle)`.

I implemented this in my repository ([see this commit](https://github.com/brauliobarahona/WEC-Sim-1/commit/e07080b8f538d74d7542450da140195b511e5a8f)). I meant to make a pull request, but there are many changes that are prbably not relevant for the oficial WecSim repository and since I did not had a welll organized branching method, I think it will be a pain to pull it if you wanted.

Anyhow, it is trivial to implement, but I personally think you can save people time if you add the shuffle option because it guarantees a sort of random initialization of `rng` when `rand` is used.

yuyihsiang, 2015-07-04:
I usually set

waves.randomPreDefined = 1,2,3,......

And modify the wecSim.m file

closed 2015-07-04 by brauliobarahona

cmichelen, 2015-07-06:
Maybe we should have an `if` statement for when `waves.randPredefined = 0` and use `shuffle` in that case. All that would mean is that the random seed could be anything except `0`. Also, like Braulio said, the description is wrong. The description implies that `waves.randPredefined' is a flag (binary 1 or 0) when in reality it is the random seed and can take any value.

kmruehl, 2015-07-09:
I changed the way waveClass runs so that when waves.randPreDefined = 0, there is a random phase (even for batch runs) where the seeded random number is 'shuffled'. When waves.randPreDefined = 1, the random phase is seeded, with a seed = 1.

        function setWavePhase(obj)                                     
        % Used by waveSetup
        % Sets the irregular wave's random phase
            if obj.randPreDefined ~= 0  
                rng(1);             % Phase seed = 1
            else 
                rng('shuffle');     % Phase seed shuffled
            end
            obj.phaseRand = 2*pi*rand(1,obj.numFreq);
            obj.phaseRand = obj.phaseRand';
        end