WEC-Sim/WEC-Sim #6: PTO Blocks Internal Mechanics Indexing Error

ratanakso, 2015-07-16:
Hello,

I just pulled the latest WEC-Sim master to work with my PTO-Sim and I have noticed that there are blocks called Assignment (eg., 5th Index) that takes the PS-Simulink Converter signal and [0 0 0 0 0 0] constant block. In other words, this block's inputs are Y0 & U and output Y. I am not sure why those blocks are there. When I ran WEC-Sim with my PTO-Sim and looked at my plots all the results were wrong. Then, when I removed all the [0 0 0 0 0 0] and Assignment blocks, and re-ran WEC-Sim that coupled with PTO-Sim again, the results are correct. Could you please look into this issue.

Thanks,
Nak

yuyihsiang, 2015-07-16:
Hi Nak

Please check with Kelley and Carlos (he is out of town) to see what was changed in the PTO-sim and why.

Thanks,

Yi-Hsiang Yu, Ph.D.
Research Engineer | Water Power Program
National Renewable Energy Laboratory (NREL)
yi-hsiang.yu@nrel.gov<mailto:Yi-Hsiang.Yu@nrel.gov> | Phone: (303) 3847143

kmruehl, 2015-07-16:
I looked at this issue with Nak, but we will need to wait until Carlos returns. I’m not sure what all of this initialization does and why it is not compatible with PTO-Sim.

| Kelley Ruehl | Water Power Technologies | Sandia National Laboratories | Phone: (505) 284.8724 | Email: kelley.ruehl@sandia.gov<mailto:kelley.ruehl@sandia.gov> |

yi-hsiang.yu@nrel.gov<mailto:Yi-Hsiang.Yu@nrel.gov<mailto:yi-hsiang.yu@nrel.gov%3cmailto:Yi-Hsiang.Yu@nrel.gov>> | Phone: (303) 3847143

ratanakso, 2015-07-17:
Hi everyone,

So, I have spent a couple hours looking into this and I would like to share what I've found. 
1) WEC-Sim without PTO-Sim: It means there is no modification to the WEC-Sim code (just straight up from WEC-Sim master).
- Therefore, [0 0 0 0 0 0] constant blocks and assignment (# index) blocks are there.
Results from running the simulation:
+ output.ptos.vel is set to a zero array (eg., 40001x1) 
+ output.ptos.pos is set to a zero array (e.g., 40001x1)
+output.ptos.forceOrTorque which is the Fpto has an amplitude of 0.884 N. Since output.ptos.vel is a zero array, I would expect Fpto to be zero as well. Need explanation.

2) The same as 1) but with [0 0 0 0 0 0] constant blocks and assignment (# index) blocks were removed.
Results from running the simulation:
+ output.ptos.vel and output.ptos.pos  agreed with relative velocity and relative position. Make sense.
+ output.pto.forceOrTorque which is the Fpto has an amplitude of 0.5436 N. Really? The math did not match. Need explanation. 

Conclusion:
1) & 2) indicate that Fpto is not right. Also, why Fpto =c*v (since k=0)? Fpto should equal to -c*v if c is defined as positive. Fpto should have opposite direction to relative velocity for generating. When Fpto has the same direction as relative velocity, it means motoring.
 
3) WEC-Sim with PTO-Sim: It means a Translational PTO UD Force (Local Z) is used and this PTO-Sim is a linear damping (no hydraulic or direct drive PTO-Sim was used). 
- Therefore, [0 0 0 0 0 0] constant blocks and assignment (# index) blocks are there.
- I define Fpto = -c*v where c is a positive value of 1.2e6
Results from running the simulation:
+output.ptos.vel is set to a zero array (eg., 40001x1) 
+output.ptos.pos is set to a zero array (e.g., 40001x1)
+Fpto is zero. Therefore, it makes sense since output.ptos.vel is also zero. Although, it is clear that [0 0 0 0 0 0] constant blocks and assignment (# index) blocks cause the results to be zero. 

4) The same as 3) but with [0 0 0 0 0 0] constant blocks and assignment (# index) blocks were removed.
-I define Fpto = -c*v where c is a positive value of 1.2e6
Results from running the simulation:
+ output.ptos.vel and output.ptos.pos  agreed with relative velocity and relative position. Make sense.
+ Fpto has an amplitude of 0.833 MN. Make sense and the math is matched. The direction of Fpto is opposite to the relative velocity and therefore generating. Which it should be. 

In conclusion, I would suggest the team to look into the issues I discussed above because I believe these issues need to be addressed. One major error that is obvious to me is the use of [0 0 0 0 0 0] constant blocks and assignment (# index) blocks.

Regards,
Nak

cmichelen, 2015-07-29:
Should be fixed. Nak please check that it is working, and if so we can go ahead and close this issue.

The responseClass PTO ouput format had to be updated.

ratanakso, 2015-07-31:
Carlos,

Thank you for looking into this issue. One more thing that needs to be fixed is that PTO force should not be in the same direction as relative velocity. For example, in RM3 model, if you go to Translational PTO (Local Z) and then look inside InternalMechanics, you will find that PTO force = k*x+c*v=c*v (since k=0). The sign should be minus since the PCC Damping Coefficient is defined as positive. For generating, we want PTO force to have opposite sign to relative velocity. This can be fixed by changing the Add block (from ++ to --).

Regards,
Nak

ratanakso, 2015-08-05:
Everything is working now. Thanks for fixing these issues. 

Regards,
Nak

closed 2015-08-05 by ratanakso