WEC-Sim/WEC-Sim #11: User Defined Force PTO block errors

ahagmuller, 2015-08-18:
Hi,
When using the user defined for PTO block (z direction for us) we got the following error:
(ahcontrol4 is our .slx file)

Caused by:
    Error using wecSim (line 130)
    Error in port widths or dimensions. Output port 1 of 'ahcontrol4/Global Reference Frame/Solver Configuration/EVAL_KEY/INPUT_DEMUX_3_0' is a one dimensional vector with 6
    elements.
    Error using wecSim (line 130)
    Error in port widths or dimensions. Input port 1 of 'ahcontrol4/Global Reference Frame/Solver Configuration/EVAL_KEY/INPUT_3_1_1' is a one dimensional vector with 1
    elements.

lawsonro3, 2015-08-18:
Hi,

Could you send me the case you are trying to run, its difficult to look into the problem from just the info above.

If you are not comfortable posting the info on this website, please email it to me at michael.lawson@nrel.gov.

cmichelen, 2015-08-18:
You should pull the latest version. I fixed an issue with these blocks about a week ago. If this doesn't solve the problem I would suggest looking at the dimensions of the inputs and outputs.

The output from the user defined block is a bus of 3 signals (position, velocity, and acceleration). The first step is to use a bus selector. Each of the 3 signals are 6x1 (6 Degrees of freedom) vector at any time-step.  

The input you need to provide is the force. This should be a scalar at any time-step.

Hope this helps,

Carlos

kmruehl, 2015-08-19:
This is related to https://github.com/WEC-Sim/WEC-Sim/issues/6, which was already addressed and closed by @cmichelen

kmruehl, 2015-09-02:
@ahagmuller did this response resolve your issue?

ahagmuller, 2015-09-11:
Hi,
Sorry for the late reply, I worked through this and found that my SIMULINK library was not updated correctly and got this part all to work now. Thanks!

Alex

kmruehl, 2015-09-14:
great! I'm glad the issue has been resolved.

closed 2015-09-14 by kmruehl