WEC-Sim/WEC-Sim #17: Surge response using 'floating' constraint

kmruehl, 2015-08-19:
I am modeling a simple cylindrical buoy in WECSim. It's a small-scale model (0.5 m diameter) subjected to 2 s period waves of amplitude 0.05 m (0.1 m total wave height). All of the hydrodynamic coefficients were obtained using NEMOH and they have been validated to be correct. I've modeled the buoy in Simulink using a floating constraint with the base tied to the seabed.

@bjrose86

The heave response is close to experiments - the heave amplitude should be 0.05 m (same as wave amplitude), yet WECSim calculates 0.04 m. Likewise, the buoy should pitch about +/- 4 degrees, yet WECSim calculates +/- 2 degrees. The discrepancy seems likely to do with the surge motion. WECSim calculates that the buoy surges with a much larger amplitude of 2.5 m (close to half the wavelength), whereas I believe the surge amplitude should be closer to the wave amplitude of 0.05 m.

Perhaps there is an issue with the surge calculation when using the floating constraint?

kmruehl, 2015-08-19:
@bjrose86 WEC-Sim's floating constraint supplies no restoring force in any DOF, and surge motion has no hydrostatic restoring force. Can you post the *.h5 file or plot the surge and heave coefficients for us to review?

bjrose86, 2015-08-20:
The above problem was due to a conversion issue from NEMOH into hdf5. I pulled the latest version of BEMIO and updated 'scale = False' in the python code. Attached are images showing the new (correct)  responses for a cylindrical buoy modeled using a floating constraint. 

[image]
[image]
[image]

bjrose86, 2015-08-20:
The correct pitch amplitude is 2.7 degrees (not 4 degrees as stated earlier).

kmruehl, 2015-08-20:
I'm glad to hear the re-running BEMIO using non-dimensional coefficients resolved the issue. 

We recently changed the WEC-Sim code so that it imports non-dimensional hydro coefficients from the BEMIO h5 file (hence 'scale = False'). The WEC-Sim code them dimensionalizes these terms during simulation. Using 'scale = True'  should only be done when the user wants dimensionalized hydro coefficients in the h5 file. This option however is NOT compatible with WEC-Sim.

closed 2015-08-20 by kmruehl