WEC-Sim/WEC-Sim #24: Output sampling rate - Enhancement request

bradling, 2015-08-26:
It would be nice to have the ability to output results at a lower sampling rate than the simulation is run at. It would reduce memory demands and increase runtime especially in accelerator and rapid accelerator mode for batch runs.

Maybe add a property to simulationClass something like dtOut?
Then the decimation of all the "to workspace" blocks could be set to "(simu.dtOut/simu.dt)"?

kmruehl, 2015-09-02:
This is a great idea, and your implementation suggestion sounds appropriate. However, this is not currently a priority for the SNL/NREL WEC-Sim team. If you would like to make this revision to the code, we encourage you to do so, and then send us a pull-request (https://help.github.com/articles/using-pull-requests/) and we will integrate your revision into WEC-Sim mater (https://github.com/WEC-Sim/WEC-Sim).

cmichelen, 2015-09-16:
@bradling completed this enhancement. Thanks!

closed 2015-09-16 by cmichelen