WEC-Sim/WEC-Sim #26: time-domain state-space realization and frequency-domain state-space realization?

wyg110, 2015-08-30:
Dear Dr.  Nathan Tom:

      Very please to contact you again and we are researchers from Shanghai Jiao Tong University.
In your paper “ Recent Additions in the Modeling Capabilities for the WEC-Sim-v1.1 Wave Energy Converter Design Tool” You mention that “Currently, WEC-Sim allows for the state-space realization of the hydrodynamic radiation coefficients either in the frequency-domain (FD) or time-domain (TD);”.  However, we do not know which WEC-Sim command we should use in order to
perform a TD state-space realization or FD state-space realization. In the  WEC-Sim-v1. 2. 1 wecSimInputFile for the RM3 tutorial, when we choose the “ Irregular Waves using BS Spectrum with State Space Calculation” in this file, there is a line command “simu.ssCalc = 1;   %Control option to use state space model”. What does “simu.ssCalc = 1” mean? Does “simu.ssCalc = 1” mean time-domain state-space realization? Or does “simu.ssCalc = 1” mean frequency-domain state-space realization? How should we set the simu.ssCalc value in order to perform a time-domain state-space realization? How should we set the simu.ssCalc value in order to perform a frequency-domain state-space realization? Or are there any other input files in your WEC-Sim-v1.2. 1 routine that we should change in order to choose between the time-domain state-space realization and frequency-domain state-space realization?
        Thanks for your consideration, and we are looking forward to hearing from you.

Sincerely yours,
wyg110

yuyihsiang, 2015-08-31:
Hi

In Nathan's paper, he included both the time-domain and frequency domain state-space realizations. However, in the current WEC-Sim V1.2.1, we only included the time-domain state-space realization because the frequency domain one requires the Signal Processing Toolbox distributed by MATLAB. 

Thanks,

Yi-Hsiang

closed 2015-08-31 by yuyihsiang

wyg110sjtu, 2016-05-21:
Hi, Yi-Hsiang

      In your paper “ Recent Additions in the Modeling Capabilities for the WEC-Sim-v1.1 Wave Energy Converter Design Tool” You mention that “Currently, WEC-Sim allows for the state-space realization of the hydrodynamic radiation coefficients  in the  time-domain (TD);”. However,  I have not  found the specific MATLAB code inside the WEC-Sim software for the time-domain (TD) state-space realization of the radiation forces. Can you please tell me in which sub-routine does the MATLAB code for the time-domain (TD) state-space realization of the radiation forces exist?  Have you utilized the MATLAB function imp2ss to perform the time-domain (TD) state-space realization of the radiation forces ?
A related more general question is : Where (in which  sub-routine of the WEC-Sim) is the
 MATLAB code for solving the WEC 6-dof equations of motion?

Sincerely yours

wyg110