WEC-Sim/WEC-Sim #32: Speed up stl file reading for non-linear hydro.

cmichelen, 2015-09-16:
Non-linear hydro requires the stl geometry to have a large amount of cells. The current stl reader function is extremely slow, limiting the size of the stl file you can use.

cmichelen, 2015-09-16:
I found a different stl-reading function online. It reads an stl file with 8000 cells in 0.07s (compared to 6+ minutes with the current function). It read an stl with 33000 cells in 0.2s (can't even read this one with the previous function, matlab crashes.

I will close this issue once I fully incorporate it and push.

yuyihsiang, 2015-09-16:
Hi Carlos

That's great.

Yi-Hsiang Yu, Ph.D.
Senior Research Engineer | Water Power Program
National Renewable Energy Laboratory (NREL)
yi-hsiang.yu@nrel.gov<mailto:Yi-Hsiang.Yu@nrel.gov> | Phone: (303) 3847143

commit b527d4a2 (2015-09-17): faster stl reader, closes #32
closed 2015-09-17 by cmichelen