WEC-Sim/WEC-Sim #34: Nonlinear hydro with irregular waves not working

cmichelen, 2015-09-22:
error message:

Size mismatch (size [1001 x 1] ~= size [96 x 1]).

Function 'Body Elements/Rigid Body/Wave Diffraction and Excitation Force Calculation/Nonlinear FK Force Variant Subsystem/With Nonlinear FroudeKrylov Force/Diffarction and Excitation Forces' (#260.3838.3842), line 102, column 16:
"k.*z"

ahagmuller, 2015-09-29:
Caused by:
    Error using wecSim (line 157)
    Errors occurred during parsing of MATLAB function 'Rigid Body/Wave Diffraction and Excitation Force
    Calculation/Nonlinear FK Force Variant Subsystem/With Nonlinear FroudeKrylov Force/Diffarction and Excitation
    Forces'
Hi,
I believe this is the same error.

    Error using wecSim (line 157)
    Error occurred in 'ah_hsp_control3/Rigid Body/Wave Diffraction and Excitation Force Calculation/Nonlinear FK
    Force Variant Subsystem/With Nonlinear FroudeKrylov Force/Diffarction and Excitation Forces'.
 
>> 
nlhydro set to 2
also mass is user defined, not "equilibrium" (not sure if this has much to do with it)

Thanks!

Alex

yuyihsiang, 2015-09-29:
Hi Alex

Please note that the nonlinear feature does not work for "user-defined" irregular wave cases and you will need a correct stl file for the body. Can you upload your "wecsimInput.m" file so that we can try to duplicate what happened?

Thanks,

Yi-Hsiang Yu, Ph.D.
Senior Research Engineer | Water Power Program
National Renewable Energy Laboratory (NREL)
yi-hsiang.yu@nrel.gov<mailto:Yi-Hsiang.Yu@nrel.gov> | Phone: (303) 3847143

cmichelen, 2015-11-04:
Not working for irregular waves.

Lines 110-140 in the non-linear FK matlab block in the library. Should probably loop through all the wave numbers (``k``), and not use the first wave (``k(1)``).

yuyihsiang, 2015-11-05:
Fixed and pushed

closed 2015-11-05 by yuyihsiang