WEC-Sim/WEC-Sim #36: buoyancy forces

bjrose86, 2015-09-24:
Perhaps this is an obvious question, but how does WEC-Sim calculate buoyancy forces? Does it use the hydrostatic stiffness matrix from the BEM (linearized about the specified draft), or does WEC-Sim use the .stl file to calculate the nonlinear buoyancy force as the draft changes.

cmichelen, 2015-09-24:
``simu.nlHydro`` lets you choose this behavior. By default it is set to zero, and uses 'the hydrostatic stiffness matrix from the BEM (linearized about the specified draft)'. 

If you set it to 1 or 2 it uses the stl file to calculate the non-linear buoyancy and Freude-Krylov forces as follows:
 
``simu.nlHydro = 1`` uses mean free surface elevation.
``simu.nlHydro = 1`` uses instantaneous free surface elevation.

@yuyihsiang @kmruehl We should explain this somewhere in the documentation.

closed 2015-09-24 by cmichelen

yuyihsiang, 2015-09-24:
Quick correction:
simu.nlHydro = 1 uses mean free surface elevation.
simu.nlHydro = 2 uses instantaneous free surface elevation.

Yi-Hsiang Yu, Ph.D.
Senior Research Engineer | Water Power Program
National Renewable Energy Laboratory (NREL)
yi-hsiang.yu@nrel.gov<mailto:Yi-Hsiang.Yu@nrel.gov> | Phone: (303) 3847143

bjrose86, 2015-09-24:
Thanks for the help Carlos and Yi-Hsiang - nonlinear buoyancy and FK forces are great features that you've added. To clarify, simu.nlHydro = 2 uses instantaneous free surface elevation at x =0?

cmichelen, 2015-09-24:
actually ``simu.nlHydro = 2``uses instantaneous free surface elevation at each cell (each triangle from you stl file).

bjrose86, 2015-09-24:
Ah got it. Thanks!