WEC-Sim/WEC-Sim #37: PTO Blocks Ordering for 6DOF Single Body Point Absorber

ahagmuller, 2015-09-24:
We are modeling a single body point absorber and have a few questions on the correct way to order the constraints. I will get some screenshots to help clarify our question soon. 

-For a single floating body point absorber, starting at the global reference frame, we used a fixed constraint, attached to a rotational PTO, attached to a Surge PTO, then attached to a Heave PTO, and finally the body of the device. We have no damping applied for the rotational and surge PTO's, and their locations are all the same (0,0,0). After running the simulation and looking at the reference frames, it appears that the rotational and surge PTOs stay at the 0,0,0 location, wheras the heave PTO moves only in the device's local Z, which is what we want and are only modeling damping with this PTO.  It is hard to pick out, but it looks like rotation is always occuring at the reference origin (0,0,0) instead of about the device CG, but we wanted to confirm this to be the case.

-As a check, we swapped the location of the rotational and surge PTO's in their ordering, so now we have the reference frame, fixed constraint,surge PTO, rotational PTO, heave PTO, and finally the body. When running the simulation changing nothing else, it appears that the device now has it's rotational PTO fixed to the device's origin at it's local coordinate system, and seems to pitch about it's CG. We think that this is the correct way to do this but want to confirm.

-We will be using a user defined force on the heave PTO, and want to confirm that power generated can be the body velocity*user input force (velocity in m/s, force in newtons) when the PTO damping constant is otherwise set to zero.

-Finally, is there any reason to use the fixed constraint in this scenario if  constraint forces aren't needed? It appears to run the same either way, and I am not sure that the constraint forces are meaningful for us.

 Again, I will try to post some screen shots of the two scenarios detailing the differences. 

Thanks,

Alex

cmichelen, 2015-09-24:
Hi Alex,

If I understand correctly, your device is attached to the seafloor. The attachment at the bottom is allowed to rotate, but the attachment at the body is fixed. Furthermore there is a PTO between the attachment and the body, allowing the distance between the body and the attachment to change. Is this correct?

If so I would use ``Global Reference Frame ==> Pitch Constraint (loc = sea floor attachment) ==> Heave PTO ==> Body``

If the body attachment can also rotate I would put a second pitch constraint right before the body.

-Carlos

ahagmuller, 2015-09-24:
Carlos,
The device is a floating single body point absorber tethered to the sea floor. The device is free to move in surge, heave and pitch. The PTO operates only in the devices local "z". 
Alex

cmichelen, 2015-09-24:
Hi Alex,
I read your question again and will try to answer them.

- In the first case you describe:
   * yes the rotation would be about [0,0,0]

cmichelen, 2015-09-24:
*- In the first case you describe:*
       - yes the rotation would be about [0,0,0]
       - *Note:* The surge constraint acts in the surge direction of the rotated frame not global surge.

*- In the second case you describe:*
       - I don't think it is rotating about the cg of the body (or the global [0,0,0]). Since the pitch constraint comes after the surge constraint I think the point of rotation is surging (global surge). That is, the surge constraint acts in the global surge, and the rotation point is attached to the the moving end point of the surge constraint. The body can then heave about the surged, and rotated frame.

*- About the user defined PTO:*
       - It provides you with the position, velocity, and acceleration of the heave PTO along its heave direction (body local z), and allows you to supply a force along that same direction. Does this answer your question?

*-About fixed constraint:*
       - No you do not need the fixed constraint
       - Also you could use pitch and surge constraints instead of PTOs with zero damping.

Hope this helps,
Carlos

ahagmuller, 2015-09-29:
Hi Carlos,
Okay, I think we have close to what we want. We offset the point that the tether passes through to the bottom of the device, and ordered the blocks Fixed Constraint, Surge, Heave, and then Pitch.
When watching the frames this looks correct. 

One question I have and this is probably obvious, is there any reason one wouldn't use the applied PTO force*pto velocity to get power when not using the constant damping in the wecsiminput file?

Thanks!

Alex

kmruehl, 2015-09-29:
Alex,

In response to your PTO question, the user-defined PTO blocks in WEC-Sim (ex: UD Force, UD Torque) are 'sensed' and 'actuated' by the SimMechanics joints. An example for the Translational PTO UD Force (Local Z) is shown below. 

[image]

The joints can either be actuated by a user-defined force, or they can have a linear damping (pto.c) or stiffness (pto.k) applied to the internal mechanics as specified in the wecSimInputFile. These joints then sense position/disp/vel, which are bussed together as the 'Response' signal

[image]

You could certainly take the applied_pto_force*sensed_pto_vel to get the output power. I do not see any downsides to this other than this power term is not included in the current WEC-Sim output structure. For example in the above example pto1_output includes powerInternalMechanics, which is pto.c*sensed_pto_velocity

-Kelley

ahagmuller, 2015-09-29:
Hi Kelley,

I ended up just making a block in that multiplied the PTO velocity by the User defined force to output power. 

In the end we are trying to achieve variable damping, which can't be done with the standard internal mechanics structure. 

I am thinking just making a new variable for a damping value that defines the user input force to the PTO block as this new damping value times the PTO velocity. I think this should be close to how it is currently functioning, and if you were to set the damping value as a constant it should be the same, with the ability to vary the damping if desired. 

Thanks for the clarification!

Alex

kmruehl, 2015-09-29:
Alex,

If you are applying variable damping what you've proposed is the correct approach. Both the user-defined force (for non-linear forcing) and the internal mechanics pto.c (for linear damping) have the same result, the nuances of how they are implemented in the joint is just slightly different due to the SimMechanics implementation. 

Does this resolve your question? Can I close this issue?

-Kelley

ahagmuller, 2015-09-29:
Kelley,
Yes it does thanks!

Alex

kmruehl, 2015-09-29:
glad we could help.
-Kelley

cmichelen, 2015-10-01:
We should add the user-defined force/torque to the output structure. Re-opened this issue and tagged as enhancement.

bradling, 2015-10-01:
Hi Carlos,

I was under the impression that the user defined PTO force is included in the output structure as "output.ptos.forceActuation", where pto(1) is a user defined pto. If my understanding is incorrect I think I will need to make modifications to my data postprocessing.

Brad

cmichelen, 2015-10-01:
Hi Brad,

You are correct. I forgot we were already outputting it.  Thanks!

closed 2015-10-01 by cmichelen