WEC-Sim/WEC-Sim #38: Buoyant force/restoring force questions

ahagmuller, 2015-09-24:
The way we understand it, restoring force is zero when the device is at static equilibrium with it's weight and the buoyant force acting on it, and when the device is displaced/submerged from this static position the greater displacement is what generates a restoring force. 

Our question then is, if we apply a user defined force, will the restoring force increase until equilibrium is reached with the force and the new displacement of the device in the model? Would running the model without any waves in still water show this?  

We are running the simulation with non-linear buoyancy now.

Thanks,

Alex

cmichelen, 2015-09-24:
Hi Alex,

The best way to test it is with no waves. First run your simulation without the external force and with no waves. The device should not move (or move just slightly if the volume and mass given do not match exactly).

Then add your external force. If the external force is constant it should oscillate towards its new equilibrium position.

Hope this helps,

Carlos

ahagmuller, 2015-09-24:
Carlos,

Thanks, this sounds how we expected.  We will give it a try. 

Regards,

Alex

closed 2015-10-01 by cmichelen