WEC-Sim/WEC-Sim #39: Natural Frequency Question

ahagmuller, 2015-10-01:
Hi,
Would it be correct to say for WEC-SIM, if you set up a simulation with no waves and started out  with an applied force to a body that was instantaneously removed, that it should oscillate at it's natural frequency?

Thanks,
Alex

kmruehl, 2015-10-01:
Yes, this is similar in concept to running a free decay test. When you initially displace the WEC it will oscillating from its initial displacement to its steady state equilibrium at its damped natural frequency. From this you can calculate the damping ratio, and thus the natural frequency of the WEC in that DOF.
-Kelley

ahagmuller, 2015-10-01:
Awesome, thanks!

Alex

ahagmuller, 2015-10-05:
Hi,
I have been running the code with the above concept, but the results seem a bit strange. Here is the body velocity with no wave on a first run with an initial force removed:

[image]

The natural damped natural period above appears to be about 4.8 seconds, but it looks like there is some wave interactions that cause the device to keep oscillating.

And here is results with the body mass set to equilibrium and no force applied and then removed:

[image]

The damped natural period appears to be about 4.4 seconds. It does not seem correct that the device would continue to oscillate with greater amplitude as though an external force is being applied. It looks the opposite of what a decay test would be. Any thoughts?

Thanks!

Alex

cmichelen, 2015-10-05:
We have seen this kind of behavior before. I believe it is a numerical instability due to very small damping value (radiation damping). If you force the radiation damping to zero (create a new h5 file), it should oscillate and never decay (nor grow). If you have large damping it decays. But sometimes when you have very small damping you see this kind of numerical instability. The best solution might be to add some extra damping (linear or quadratic).
@yuyihsiang any other ideas?

bjrose86, 2015-10-05:
Or perhaps trying hitting your device with a larger initial force to see if you get a proper decay before hitting these numerical instabilities.

ahagmuller, 2015-10-05:
Hi,
I have added quadratic damping, no linear damping and I am running
non-linear buoyancy as well. I have not tried it without non-linear
buoyancy.

I will try as you suggested. Even if it numerically unstable, would the
period be correct?

Thanks,

Alex

ahagmuller, 2015-10-05:
It is an interesting behavior, it looks as though the waves produced by the
device are reflected back somehow to the device at resonance. Probably not
what is happening though.

ahagmuller, 2015-10-07:
Below is a screen shot of the velocity after increasing the linear damping in heave.I did not have linear damping before, only quadratic damping. This is with an initial offset displacement of 2.5m in the bodyclasss.m file, using non linear hydro, and user defined mass. You can see it initially decays rapidly, but then has a periodic response every 60 seconds or so that also decays. This still seems strange to have this secondary response, but the natural damped period does not seem to have changed appreciably from the first test with no linear damping applied. What do you guys think?
Also, I have not applied a large force initially as suggested yet, this is just with a large initial displacement. Thanks!

Alex

[image]

ahagmuller, 2015-10-07:
In this simulation the mass is set to equilibrium with non linear hydro turned off, and an initial displacement of 2.5m. Linear damping is added for in the first 3 elements of the vector. It looks like there might be some kind of "step" in the geometry of the device, since it initially settles quickly to an equilibrium position, but then for some reason every 60 seconds the equilibrium position changes causing a response in the system, which then settles to a slightly different new position. Since non-linear hydro is eliminated, it shouldn't be caused by the resolution of the .stl file, could this be the resolution of the original NEMOH mesh causing this?

Thanks for your input!

Alex

[image]

cmichelen, 2015-10-07:
Are you using ``waves.type = 'noWaveCIC';``? The default convolution integral time is set to 60s. You should take a look at your impulse response function in the h5 file, and select a convolution  integral time that looks like it has decayed already. You can change this with ``simu.CITime = 60;``. in your input file (change 60 to something else).

ahagmuller, 2015-10-07:
Hi Carlos,
This sounds like what we are seeing with the periodic impulse at 60 seconds. We will give this a try. It seems like if this was the case , it would have an effect on our simulation as well. Would this be the case?

Thanks!

Alex

cmichelen, 2015-10-07:
It would have an effect in any simulation that uses convolution integral (regular waves you can run with constant coefficients instead of conv integral). But I am not sure how big an effect it would be. Other force (e.g. excitation) might be much larger than this effect.

ahagmuller, 2015-10-07:
Got it. Thanks I will report back. 

Alex

ahagmuller, 2015-10-08:
Alright, changed the CIC time to 190 seconds (200 seconds is the maximum i can do), and removed the additional linear damping. The start position offset is 10m. The period appears to be 4.4 seconds. At the end of 190 seconds, I get an excitation force again. Does this seem okay?

Thanks,

Alex

[image]

cmichelen, 2015-10-12:
That behavior at the end is definitely not physical. Since it happens at 190s it has to be related to the CITime. Can you share a plot of your irf? 
This is what it looks like for the float in the RM3 tutorial:

``plot(body(1).hydroData.hydro_coeffs.radiation_damping.impulse_response_fun.t,body(1).hydroData.hydro_coeffs.radiation_damping.impulse_response_fun.K(3,3,:))``

[image]

closed 2015-11-02 by cmichelen