WEC-Sim/WEC-Sim #40: OSWEC: State Space for Irregular

ratanakso, 2015-10-01:


ratanakso, 2015-10-01:
Hello,

Can we run state space with the OSWEC example in irregular wave case?

Regards,
Nak

yuyihsiang, 2015-10-01:
We did not include state space in the OSWEC example because the hydrodynamic coefficients for the base are not good, particularly for some DOFs, where the base is trying to oscillating when it is sitting on the sea bed. The hydrodynamic coefficients for the base were not used in WEC-Sim, since the base was fixed. However, the bad coefficients will cause BEMIO to error when creating the state space coefficients for the base.
If you really want to model it in OSWEC example, I think the following steps will work:
1. Run BEMIO to output the h5 format for body 1 only with state space coefficients.
2. Remove the body 2 settings in the wecsimInput.m file
3. Remove body 2 in OSWEC.slx and connect the rotational PTO joint directly to the seabed.

I think this should work, but I have never tried that. Let me know if you have any question.

-- Yi-Hsiang

ratanakso, 2015-10-02:
Yi-Hsiang,

Thank you. I don't need to run state space in the OSWEC example yet, but if I do, I will follow your instructions.

Regards,
Nak

closed 2015-10-06 by cmichelen