WEC-Sim/WEC-Sim #41: OSWEC: Regular Wave Case

ratanakso, 2015-10-05:
Hello,

It appears to me that the default OSWEC tutorial example case does not work for all wave period cases. For example, it failed when I tried to run at waves.T=16 at waves.H=2.5.

Below is the PTO information:
%% PTO and Constraint Parameters
constraint(1)= constraintClass('Constraint1'); % Initialize ConstraintClass for Constraint1
constraint(1).loc = [0 0 -10];

pto(1) = ptoClass('PTO1');                     % Initialize ptoClass for PTO1
pto(1).k = 0;                                  % PTO Stiffness Coeff [Nm/rad]
pto(1).c = 0;                                  % PTO Damping Coeff [Nsm/rad]
pto(1).loc = [0 0 -8.9];                       % PTO Global Location [m]

Here is the error message:
Error using wecSim (line 157)
Derivative of state 'Rotational_PTO_Local_RY.Pitch_PCC_Revolute_Joint.Rz.w' in block 'OSWEC/Rotational PTO (Local RY)/Pitch PCC
(Revolute Joint)' at time 65.95 is not finite. The simulation will be stopped. There may be a singularity in the solution.  If
not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances)

So, I changed the time step to 0.01s (simu.dt = 0.01) and it ran but results seem wrong. For example, when I look at output.bodies(1).position(:,5) which represents pitch (Ry), the angle is around +/- 75 degrees (max at +/-85 degrees).

Regards,
Nak

cmichelen, 2015-10-05:
It might just be near resonance. If it is near resonance and you have no viscous drag, the response can be unrealistically large. You can check this by running it  at different periods and plotting the RAO of the pitch response.

ratanakso, 2015-10-05:
Hi Carlos,

RAO = Relative motion [m]/Wave Height [m] for a point absorber

I am not sure how to define RAO for OSWEC

Assuming RAOs are defined as the following:
For X-axis:
RAO=motion in X-axis [m]/Wave Height [m]
For Pitch:
RAO=Pitch [rad]/Wave Height [m]

For waves.H=2.5 and simu.dt = 0.1s
RAO(waves.T=6) = 0.45 (X-axis) unit [m/m]          RAO(waves.T=6) = 0.1 (Pitch) unit [rad/m]
RAO(waves.T=7)=0.57 (X-axis) unit [m/m]            RAO(waves.T=7)=0.12 (Pitch) unit [rad/m]
RAO(waves.T=8)=0.72 (X-axis) unit [m/m]            RAO(waves.T=8)=0.15 (Pitch) unit [rad/m]
RAO(waves.T=9)=0.89 (X-axis) unit [m/m]            RAO(waves.T=9)=0.18 (Pitch) unit [rad/m]
RAO(waves.T=10)=1.13 (X-axis) unit [m/m]          RAO(waves.T=10)=0.24 (Pitch) unit [rad/m]

Started to fail, so I changed simu.dt to 0.01s
RAO(waves.T=11)=2 (X-axis) unit [m/m]               RAO(waves.T=11) = 0.63 (Pitch) unit [rad/m]
RAO(waves.T=12)=2 (X-axis) unit [m/m]               RAO(waves.T=12) = 0.63 (Pitch) unit [rad/m]

Starting from waves.T=11 it appears to fail. 

For RM3 we can incorporate cd and characteristicArea as the following:
% Spar
r2 = 15;  % plate radius
A2 = pi*r2^2;
body(2).cd = [0 0 4.3 0 0 0];
body(2).characteristicArea = [0 0 A2 0 0 0];

% Float
rspar = 3;     % spar radius
r1 = 10 - rspar; % float radius is 10-rspar
A1 = pi*r1^2;
body(1).cd = [0 0 1.4 0 0 0];
body(1).characteristicArea = [0 0 A1 0 0 0];

How about OSWEC?

cmichelen, 2015-10-06:
Hey Nak,

I think it is just resonance. I used linear damping and increased it until it was stable. This is what I used:

``body(1).linearDamping = [0, 0, 0, 0, 1*10^7, 0];``

and these are the results:

[image]

closed 2015-10-06 by cmichelen