WEC-Sim/WEC-Sim #42: Hydrodynamic Coefficients causing error

Arven, 2015-10-07:
I have succeeded in running the RM3 tutorial case in Wec-Sim, using the provided rm3.h5 file for the hydrodynamic coefficients.

However, when I tried to convert the provided rm3.out file into an rm3.h5 using BEMIO and the readWamit script, I encountered a problem. The file seemed to convert without issue, but when I try and run Wec-Sim with this converted file, it fails and throws up an error message (Enclosed at end of post).

I am using a version of BEMIO last updated in June, I think. I see that the Readme file mentions that the rm3.out and rm3.h5 are based on different runs. Has this problem been noticed before? I concede I may be making an error during the conversion.

Any help or info would be greatly appreciated.

Error message as follows:
'Index exceeds matrix dimensions.

Error in interp1 (line 125)
        V = V(idx,:);

Error in bodyClass/regExcitation (line 159)
                obj.hydroForce.fExt.re(ii) =
                interp1(obj.hydroData.simulation_parameters.w,squeeze(obj.hydroData.hydro_coeffs.excitation.re(ii,1,:)),w,'spline');
                
Error in bodyClass/hydroForcePre (line 91)
                    obj.regExcitation(w);

Error in wecSim (line 55)
    body(kk).hydroForcePre(waves.w,simu.CIkt,waves.numFreq,simu.dt,simu.rho,waves.type,waves.waveAmpTime,kk,simu.numWecBodies,simu.ssCalc);'

lawsonro3, 2015-10-07:
Hi Arven,

I can take a look at this this morning, but it would help a ton if you could zip the files you are using and post them on this site (preferred) or via dropbox. Its't difficult to tell exactly whats wrong exclusively from your error messages.

Thanks!

-Mike

Arven, 2015-10-07:
Thanks Mike.

I'm sharing a Dropbox link as I had difficulty uploading some of the files using the website.

https://www.dropbox.com/sh/5zcz76g7fwl9h1g/AAD0QskhygbKXJV8HIxX0PDqa?dl=0

Most of the included files are unchanged from the initial time I downloaded Wec-Sim. A brief description follows.
rm3_Original.h5 : The original rm3.h5 file included with Wec-Sim. This file works for me.
rm3.out : The original rm3.out file included with Wec-Sim.
rm3.h5 : The h5 format file I created using BEMIO and the included readWamit.py file. (When I use this file for Hyrdo Data in wecSimInputFile.m, I get the error message)
wecSimInputFile.m : My input file, with minor modifications.

If you require any more information, please let me know. In the meantime, I'll pull down the current BEMIO files and see if that makes a difference.

James

lawsonro3, 2015-11-06:
Hi James,

Sorry for the very overdue response to this question - I assume that you have already figured out this problem given that you have not re-posted - but I figured I would give you a response either way.

I downloaded you files and was able to run the case in bemio and WEC-Sim. However, it seems that you are using a very old version of bemio and WEC-Sim. Both codes are under active development, so I would make sure that you update to the latest version of both. Note that although the codes are very similar to the version you were using, some of the user inputs have changes.

Please let me know if you have any further problems.

-Mike

closed 2015-11-06 by lawsonro3