WEC-Sim/WEC-Sim #43: Mooring Stiffness Matrix Question

ahagmuller, 2015-10-08:
Hi,
What are the units for the quasi-static mooring stiffness matrix? 
Thanks,
Alex

ahagmuller, 2015-10-08:
Also, it says to use a 6x6 matrix, a lot of mooring programs output a 3x3 matrix, for example;

[image]

Do you have an example of how to use or convert a 3x3 stiffness matrix for use in WEC-SIM?

Thanks,
Alex

yuyihsiang, 2015-10-08:
Hi Alex

The mooring stiffness given in WEC-Sim represents the stiffness matrix at CG due to mooring. This means that if the mooring is connected to a point other than the CG of the body, you will have to convert it to the corresponding mooring stiffness at CG. Unfortunately, we don't have any example on how to convert that. Generally, if the mooring design is slack, you can use the 3X3 matrix directly and leave the rotational part as zeros under most operational wave conditions.  This depends on how important the mooring design is to your system.

Thanks,

Yi-Hsiang Yu, Ph.D.
Senior Research Engineer | Water Power Program
National Renewable Energy Laboratory (NREL)
yi-hsiang.yu@nrel.gov<mailto:Yi-Hsiang.Yu@nrel.gov> | Phone: (303) 3847143

[image]<https://cloud.githubusercontent.com/assets/13394872/10368008/25202a22-6d88-11e5-93a1-614d727c1518.png>

ahagmuller, 2015-10-09:
Hi Yi-Hsiang,
Is there a good example that you know of showing the elements in the 6x6 matrix? 
Thanks,
Alex

ahagmuller, 2015-10-10:
Hi,
Okay,  I am not sure how correct this is but it is the best I have now.
Given the 3x3 stiffness matrix:

[11 12 13
21 22 23
31 32 33]

Here is the structure for a 6x6:

                                                  [11 12 00 00 00 13
                                                   21 22 00 00 00 23
                                                   00 00 00 00 00 00
                                                   00 00 00 00 00 00
                                                   00 00 00 00 00 00
                                                   31 32 00 00 00 33]

Units in meters, metric tonnes, and radians from the mooring program I used.
What are the units for the stiffness matrix? I am assuming it would be meters, newtons, and radians.

Thanks!

Alex

yuyihsiang, 2015-10-13:
Hi Alex

What you did look fine to me. However, just keep in mind that if you are using the Floating constraint, it is a 3DOF (surge, heave and pitch) floating joint, which means the mooring stiffness you put in in the sway and yaw directions is useless. If you really want to use 6-DOF joint given under SimScape/SimMechanics/SecondGeneration/Joints/.

--Yi-Hsiang

closed 2015-10-28 by yuyihsiang