WEC-Sim/WEC-Sim #44: nonlinear FK force implementation

bjrose86, 2015-10-08:
I have a few general questions regarding how the nonlinear FK force is calculated. 

1. Is the motion of the body (e.g. heaving or pitching) considered when calculating the potential flow field and integrating the surface pressure? 
2. Or do you use quasi-steady approximation, that is, for each time step you solve for the potential flow field about a stationary body?
3. If you use option 2, have you done any investigation into how good this approximation is? My guess is that it would be decent as long as the body oscillation frequency is slow compared to the wave frequency. However, this stands in contrast to most WEC designs, which are tuned to the wave frequency.

I would be interested in your thoughts.

Thanks!
Brian

yuyihsiang, 2015-10-08:
Hi Brain

If the nonlinear model is used (simu.nlHydro = 2), it calculates the FK force and the bounce force based on the instantaneous position of the body and wave elevation and integrates the corresponding static and dynamic components of the pressure field from each submerged panel on the body surface.

Thanks,

Yi-Hsiang Yu, Ph.D.
Senior Research Engineer | Water Power Program
National Renewable Energy Laboratory (NREL)
yi-hsiang.yu@nrel.gov<mailto:Yi-Hsiang.Yu@nrel.gov> | Phone: (303) 3847143

bjrose86, 2015-10-08:
Thanks Yi-Hsiang - 

That makes sense. But if the body is moving, wouldn't there be a dynamic pressure in addition to the dynamic pressure from the incoming wave field (assuming a stationary body)? 

Or phrased another way - If the body is stationary, the boundary condition at each panel is that the normal velocity is zero (no-flux). If the body is moving, then the boundary condition is that the fluid normal velocity is equal to the body normal velocity. 

So, my question is, does the FK calculation consider the velocity of each panel, or just the position?

Let me know if I'm unclear, and I'd be happy to rephrase!

Thanks!

yuyihsiang, 2015-10-13:
Hi Brain

For FK force, it is directly calculated from the pressure of the flow field based on wave theory based on the instantaneous body position and free surface. However, the diffraction part is still calculated based on linear theory, which assumes the body does not move. This means the dynamic pressure due to the change of the flow field is directly represented in the nonlinear FK force. However, the dynamic pressure due to the instantaneous body position and the body motion is not included. This is just the assumption given in this type of method, which is just a weakly nonlinear approach. 

Thanks,

--Yi-Hsiang

bjrose86, 2015-10-13:
Thanks Yi-Hsiang. That was a helpful explanation of the modeling approach. 

So, if nonlinear hydro is used (simu.nlHydro = 2), then WECsim calculates the FK force internally and ignores the FK load component specified by the BEM? However, WECsim will use the diffraction force specified by the BEM?

yuyihsiang, 2015-10-13:
That is correct.

Thanks,

closed 2015-10-13 by yuyihsiang