WEC-Sim/WEC-Sim #46: Trouble running RM3-Tutorial

MalteTwo, 2015-10-14:
Hi folks,
I´m completely new to WEC-Sim. I started by trying to run the RM3-example described in the user manual, but while WEC-Sim was runnig, I got this error in matlab: 

Error using wecSim (line 97)
mxIsNumeric(prhs[0]) && mxGetLength(prhs[0])==1

Is anyone able to help me with this? I exactly followed the instructions given in the user manual, and I have no clue how to fix this. I´d very much appreciate any support! Thanks a lot!!
Malte

cmichelen, 2015-10-22:
Hi Malte,
I haven't seen that error before, but it sounds like it might be a compatibility issue. A few questions:
* Are you using the latest version of WEC-Sim?
* Which version of Matlab are you using? It has to be 2014b or above. 
* Do you have all the required packages? You can type ``ver`` in Matlab to check which packages you have. You need Matlab, Simulink, Simscape, and SimMechanics.

MalteTwo, 2015-10-23:
Dear Carlos,
thanks a lot for responding! I use WEC-Sim 1.1, Matlab 2014b and I do
have all the packages required. Should I try by updating to WEC-Sim 1.2?

Regards 
Malte

>>> Carlos Michelen <notifications@github.com> 22.10.15 15.52 Uhr >>>
Hi Malte,
 I haven't seen that error before, but it sounds like it might be a
compatibility issue. A few questions:
  Are you using the latest version of WEC-Sim? Which version of Matlab
are you using? It has to be 2014b or above.  Do you have all the
required packages? You can type ver in Matlab to check which packages
you have. You need Matlab, Simulink, Simscape, and SimMechanics.

yuyihsiang, 2015-10-23:
Hi Malte

I would strongly suggest you update to WEC-Sim version 1.2.1 as well as the latest BEMIO, since we  have changed the way how we normalize all the terms when we solve the equation of motion in WEC-Sim. If the WEC-Sim is not used with the correct BEMIO version, the code will most likely crash. 

--Yi-Hsiang

MalteTwo, 2015-10-24:
Hi,
thanks a lot, I´ll try that. 
All the best
Malte

>>> yuyihsiang <notifications@github.com> 23.10.15 17.40 Uhr >>>
Hi Malte
 I would strongly suggest you update to WEC-Sim version 1.2.1 as well as
the latest BEMIO, since we  have changed the way how we normalize all
the terms when we solve the equation of motion in WEC-Sim. If the
WEC-Sim is not used with the correct BEMIO version, the code will most
likely crash. 
 --Yi-Hsiang

closed 2015-11-02 by cmichelen