WEC-Sim/WEC-Sim #48: Wec

ThommyBee, 2015-10-24:


ThommyBee, 2015-10-24:
Hi wecSim community!

I tried to get a wec model working which uses mechanical PTOs (Force = velocity*linear damping coeff.) and is to some degree comparable to the OSWEC Tutorial with an adjustable rod mounted to the seafloor.

Lets say I have a simple, sphere point absorber with radius R and [x y z] = [0 0 0] floating on the water surface, being half submerged. Waterdepth is h. Now, I would like to attach one adjustable PTO (spring damper couple) which absorbs energy out of 1. the heave motion 2. the surge motion and 3. the pitch motion. 

Obviously, the PTO can only absorb energy out of the pitch motion if it is installed on the outer hull of the sphere, so the connection point of the adjustable rod would for instance be at [x y z] = [R 0 0]. In addition, the PTO is connected to the sea floor with an offset in both y and z direction.
Moreover, there is a second PTO of the same kind attached to the point absorber at [x y z] = [-R 0 0]. A schematics of the setup which I would like to implement is attached to this post.

[image]

Although it is close to the example I am having a hard time implementing this example into wecSim. In the OSWEC Tutorial solely the rotation from the pitch motion around the flap joint is contributing to the power takeoff. Now, I do not only have the pitch motion but also the heave and the surge motion. Thus, at first I have the following (also general) questions:

1.	The pto(i).loc = [x y z] definition must be used to provide wecSim the information where the pto forces attack on the absorber body, right? Thus, pto(i).loc = [(-)R 0 0] in the proposed case!?

2.	Obviously, I see no option in including three standard, separated PTO’s from the library (Global Reference Frame -> Transl. PTO (z) -> Transl. PTO (x) -> Rotational PTO (ry) -> rigid absorber body), or is there? If I would use multiple PTO blocks in my Simulink model, how can I, after all, tell wecSim that the damping and spring characteristics have to be “shared “ between all of them?

3.	Is it even possible to use the “Rotary to adjustable rod block” which is already in the simuLink library for this special case? Would I have to adjust it to also consider the velocity from the heave and surge motion?! If so, do you have a suggestion on how to start on this? Would I have to project all occurring velocities with a non-constant angle of attack of the pto to match the vector of the pto rod?

4.	How can I use the rotary to adjustable rod block in my pto simulation block if I do not want to include a direct linear drive or hydraulic piston, but rather a simple spring/damper with defined k and c? Even if I do not want it, I have to set a “pto.SimClass” in my ptoSimInputFile such as ‘Direct Drive Linear Generator’, don’t I?

5.	In a more general way, what do the parameters ptosim.motionMechanism.crank and offset do?

I am very thankful for each and every help/hint I can get. Thanks a lot

cmichelen, 2015-10-28:
One way to do this is:
``Global reference => Fixed Constraint (loc=bottom pto attachment) => Pitch constraint (same loc) => Heave PTO => Pitch constraint (loc=top pto attachment)=>Body Block``

It should be much easier with the PTO-Sim block, but I haven't used it myself. 

@ratanakso do you have any comments on how to do it with PTO-Sim?

ratanakso, 2015-10-28:
Hello

I will take a look at your questions but for now I have attached the image below to help you understand how each parameter means. 

[image]

ThommyBee, 2015-10-29:
Thanks to both of you for the responses.

@cmichelen 
I implemented one side (let’s say right PTO) of the model in wecSin in the same way you were suggesting and actually the response of the system looks fine.
However, when connecting the second (left) side of the system in the same way (with the same line of constraints but opposite positioning) I receive an error telling me that the Revolute Joint (Pitch Constraint) has a degenerate mass distribution on its follower side. 
I assume this is due to the fact, that I cannot restrict the motion of the rigid body (absorber) twice in the same way. I tried to lower the degrees of freedom for the second added PTO by getting rid of the pitch joints and instead used float joints. This wasn’t successful. Maybe you guys know a solution?

@ratanakso 
Thanks for the schemes. At first, the case above seems to be just a rotated version of the OSWEC. However, the OSWEC still does not take heave & pitch into account.
I (next to testing the proposed method of cmichelen) tried to implement 3 UD PTOs (Pitch, Heave, Surge) and to combine the responses in my PTO simulation. Using the angular pos from the UD Pitch PTO and a significant amount of geometry and sines to derive the total displacement and velocity along the PTO Rod, then calculate the counter force from the PTO using a simple spring/damper system, and then splitting this force vector into components again using geometry and sines.
Ultimately I feed the UD PTOs with the related force (see attached scheme).. It seems that I still have some bugs in the Simulink model and thus can’t even run it.
However, does this sound like the right way to go?

[image]

cmichelen, 2015-10-29:
1. The ``degenerate mass distribution on its follower side`` can occur for several reasons. Sometimes it means that your model is over-constrained. But sometime it is some other error in your model (I have seen it happen because of bad coefficients...) 

2. I looked into using the PTO-Sim  ``Rotary to Linear Adjustable Rod`` and looked at the tutorial: ``tutorials/PTO-Sim_Tutorials/OSWEC_w_Mechanical_PTO``. But this will not work in your case because you do not have a fixed ``crank`` (radius of rotation).

3. I think your two options are the two that you have tried: 
* (i) the way I originally suggested
* (ii) the equivalent system you derived
When placing PTOs or constraints in series like you did, sometimes the order does matter. You just have to think carefully about it.

ratanakso, 2015-11-02:
I agree with cmichelen. Because you don't have a fixed ``crank`` (radius of rotation), you need to work out the math and geometry for your case. Also, it seems difficult to debug when there are multiple PTOs were used.

cmichelen, 2015-11-02:
The first method discussed would have been wrong. This is what the coordinate frames look like. The PTO is shown in the red circle. As you can see it acts on the vertical direction rather than in the direction between the two connection points.
[image]

I modified the pitch constraint so that you can specify an initial angle. This is specified in the input file with ``constraint.initiDisp``. This is what it looks like:
[image]

cmichelen, 2015-11-02:
```matlab
%% Simulation Data
simu = simulationClass();               
simu.simMechanicsFile = 'RM3.slx';      
simu.endTime=400;                       
simu.rampT = 100;                       
simu.b2b = 0;

%% Wave Information
waves = waveClass('regular');        
waves.H = 5;                          
waves.T = 8;                            

%% Body Data
body(1) = bodyClass('hydroData/rm3.h5',1);      
body(1).mass = 'equilibrium';                   
body(1).momOfInertia = [20907301 21306090.66 37085481.11]; 
body(1).geometryFile = 'geometry/float.stl';

body(2) = bodyClass('hydroData/rm3.h5',2);
body(2).mass = 'equilibrium';                   
body(2).momOfInertia = [20907301 21306090.66 37085481.11]; 
body(2).geometryFile = 'geometry/float.stl';

%% Right PTO and Constraint Parameters
top = [10,0,0];
bot = [20, 0, -10];
midx = top(1) + ((bot(1)-top(1))/2);
midz = top(3) + ((bot(3)-top(3))/2);
ang = 0;%-45*pi/180;
%mid = [midx*cos(-ang)+midz*sin(-ang), 0, -midx*sin(-ang)+midz*cos(-ang)]; 
mid = [midx, 0, midz];

constraint(1) = constraintClass('BottomConstraintR')
constraint(1).loc = bot;
constraint(1).initDisp = ang;

constraint(2) = constraintClass('TopConstraintR')
constraint(2).loc = top;
constraint(2).initDisp = ang;

pto(1) = ptoClass('PTOR');                      
pto(1).k=0;  
pto(1).c=1200000;  
pto(1).loc = mid;  

%% Left PTO and Constraint Parameters
top = [-10,0,0];
bot = [-20, 0, -10];
ang = 0;%45*pi/180;
midx = top(1) + ((bot(1)-top(1))/2);
midz = top(3) + ((bot(3)-top(3))/2);
%mid = [midx*cos(-ang)+midz*sin(-ang), 0, -midx*sin(-ang)+midz*cos(-ang)];
mid = [midx, 0, midz];

constraint(3) = constraintClass('BottomConstraintL')
constraint(3).loc = bot;
constraint(3).initDisp = ang;

constraint(4) = constraintClass('TopConstraintL')
constraint(4).loc = top;
constraint(4).initDisp = ang;

pto(2) = ptoClass('PTOL');                      
pto(2).k=0;  
pto(2).c=1200000;  
pto(2).loc = mid;  

clear top bot midx midz mid 
```

cmichelen, 2015-11-02:
I also get an error when trying to run this:

```
Error using wecSim (line 178)
* ['RM3/Global Reference Frame/Solver
Configuration']: Error evaluating equations at time 0.0.  Stopping simulation.  There
may be a singularity in the solution.  If not, try reducing the step size (either by
reducing the fixed step size or by tightening the error tolerances)

Caused by:
    Error using wecSim (line 178)
    * ['RM3/Translational PTO  (Local Z)1/Translational PCC (Prismatic Joint)']:
    'RM3/Translational PTO  (Local Z)1/Translational PCC (Prismatic Joint)' has a
    degenerate mass distribution on its follower side.
```

This example is a closed kinematic loop, which can be tricky. But SimMechanics is supposed to be able to solve it. See this example: http://www.mathworks.com/help/physmod/sm/ug/model-four-bar.html

One workaround is to put a small mass at the location where it is complaining (in this case on the follower side of 'Translational PTO (Local Z)1', like so:
[image]

I did this with a 1kg mass and it runs. You need to chose a small enough mass that it won't change your dynamics, but not so small that it causes numerical issues. You might want to put the same mass on both sides just for symmetry.

closed 2015-11-02 by cmichelen