WEC-Sim/WEC-Sim #49: Installing python packages and bemio

MalteTwo, 2015-10-25:
Hi folks,
i´m desperately trying to find a possibility to use WEC-Sim. As I figured, I need Nemoh to get hydrodynamic coefficients, which I then have to trynsform using Bemio, so I can use them as input for WEC-Sim. I managed to set up Nemoh, but now I´m struggling. 

1. I downloaded Python 3.4.3 and .whl files of the required packages (matplotlib, numpy, scipy, progressbar2, astropy, h5py). Trying to install the packages with the command "pip install matplotlib" I receive nothing but error messages. I tried from both python console and shell (SyntaxError: invalid syntax), as well as from the windows command prompt (doesn´t know commands "pip" or "python"). I can´t fix it. 
2. Trying to install bemio in command prompt, I get the same error (it doesn´t know command "python"), despite exactly following the instructions given on http://wec-sim.github.io/bemio/installing.html#downloading-bemio 

I´m really not a computer expert, so I hope someone might be able to help me learn about all this - its exciting, but also frustrating at times. Any help is highly appreciated - thanks a lot! Regards
Malte

lawsonro3, 2015-10-27:
Hi Malte,

The problem is that you have not added python to your system path. If you google around you can easily find out how to do this.

Alternatively, the easier way to get python and the required packages installed is to download and install the Anaconda Python distribution - http://docs.continuum.io/anaconda/install.

Hope that helps and let me know if you have any other questions.

-Mike

MalteTwo, 2015-10-29:
Hi Mike,
thank you very much for your help. I have now downloaded and installed
the Anaconda Python distributionn and added it to my system path. So you
solved my first problem: I have all the required packages now - great. 
Still, the installation of bemio won´t work. Running the command "python
setup.py install --user" from the command prompt, I now get the
following error message: 

Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from bemio.__version__ import base
  File "C:\.....\bemio-master\bemio\__init__.py", line 1, in <module>
    from __version__ import base, full
ImportError: No module named '__version__'

Now, what happens here? I checked, the file  '__version__.py' is there,
in the directory "....\bemio-master\bemio", right next to '__init__.py'.
 As I understand the install.py-script, it should navigate to the
folders and find the modules - I didn´t change the structure of folders
or files in the bemio-folder after installing it. Do you have an idea
how I can solve this? Again - I´d be gratefull for any kind of support!
Thanks a lot, regards
Malte

>>> Michael Lawson <notifications@github.com> 27.10.15 18.16 Uhr >>>
Hi Malte,
 The problem is that you have not added python to your system path. If
you google around you can easily find out how to do this.
 Alternatively, the easier way to get python and the required packages
installed is to download and install the Anaconda Python distribution -
http://docs.continuum.io/anaconda/install.
 Hope that helps and let me know if you have any other questions.
 -Mike

MalteTwo, 2015-11-02:
Hi Mike,
Do you think it would be possible to give me some hint on how to install
bemio? I´m desperate! Thanks a million!
Malte

Hi Mike,
thank you very much for your help. I have now downloaded and installed
the Anaconda Python distributionn and added it to my system path. So you
solved my first problem: I have all the required packages now - great. 
Still, the installation of bemio won´t work. Running the command "python
setup.py install --user" from the command prompt, I now get the
following error message: 

Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from bemio.__version__ import base
  File "C:\.....\bemio-master\bemio\__init__.py", line 1, in <module>
    from __version__ import base, full
ImportError: No module named '__version__'

Now, what happens here? I checked, the file  '__version__.py' is there,
in the directory "....\bemio-master\bemio", right next to '__init__.py'.
 As I understand the install.py-script, it should navigate to the
folders and find the modules - I didn´t change the structure of folders
or files in the bemio-folder after installing it. Do you have an idea
how I can solve this? Again - I´d be gratefull for any kind of support!
Thanks a lot, regards
Malte

>>> Michael Lawson <notifications@github.com> 27.10.15 18.16 Uhr >>>
Hi Malte,
 The problem is that you have not added python to your system path. If
you google around you can easily find out how to do this.
 Alternatively, the easier way to get python and the required packages
installed is to download and install the Anaconda Python distribution -
http://docs.continuum.io/anaconda/install.
 Hope that helps and let me know if you have any other questions.
 -Mike

lawsonro3, 2015-11-06:
Hi Malte,

Sorry for the late reply.

There is a (reasonably) comprehensive description of how to install and test bemio in the [bemio documentation](http://wec-sim.github.io/bemio/). The documentation should explicitly answer you questions, but please re-post in this forum if you are still having any problems.

-Mike

closed 2015-11-06 by lawsonro3

MalteTwo, 2015-11-06:
Hi Mike,
thanks for the response! I know anout the description in the bemio
documentation, and I follow it exactly. That is my problem: I do just
what it says, but it still doesn´t work. That is why I asked for help in
the first place. I re-posted my issue, as you suggested. Thanks again -
I really hope we can figure it out!
Regards
Malte

>>> Michael Lawson <notifications@github.com> 06.11.15 18.05 Uhr >>>
Hi Malte,
 Sorry for the late reply.
 There is a (reasonably) comprehensive description of how to install and
test bemio in the bemio documentation. The documentation should
explicitly answer you questions, but please re-post in this forum if you
are still having any problems.
 -Mike