WEC-Sim/WEC-Sim #51: Visualisation with Paraview

MattSharman, 2015-10-27:
Hi All, 

Has anyone any advice for creating STL files for visualisation in Paraview?

I'm using geometry created in Solidworks at the moment but in running with simu.paraview = 1 I'm getting this error:

>52  if (simu.nlHydro >0) || (simu.paraview == 1)
>Subscript indices must either be real positive integers or
>logicals.
>
>Error in import_stl_fast (line 77)
>if (C{4}(end)== NaN)
>
>Error in bodyClass/bodyGeo (line 177)
>                [obj.bodyGeometry.vertex, obj.bodyGeometry.face,
>                ~] = import_stl_fast(fname,1,2);
>
>Error in wecSim (line 54)
>        body(kk).bodyGeo(body(kk).geometryFile)

I think it might have something to do with my STL file, not sure, Any ideas?

Many thanks, 
Matt

cmichelen, 2015-10-28:
It might be your STL file. I was having trouble with STL files created in different machines (Winodws vs Linux), but I thought I had fixed that.

Can you try running the RM3 example with simu.paraview=1? That one should work. Also can you share an STL file that gives this error?

Thanks,

Carlos

MattSharman, 2015-10-30:
Hi Carlos, 

Here's a link to the .STL: https://www.dropbox.com/s/47l99pfmu6kq3cp/Float.STL?dl=0

Trying the RM3 i get a little further then get this error message:

>Error using fprintf
>Invalid file identifier. Use fopen to generate a valid file
>identifier.
>
>Error in bodyClass/write_paraview_vtp (line 434)
>                fprintf(fid, '<?xml version="1.0"?>\n');
>
>Error in wecSim (line 262)
>        body(ii).write_paraview_vtp(output.bodies(ii).time,
>        output.bodies(ii).position, bodyname,
>        simu.simMechanicsFile, datestr(simu.simulationDate),
>        cellareas, hspressure, wpressurenl, wpressurel);

Many thanks for your help!
Matt

cmichelen, 2015-11-02:
Hi Matt,

I cannot open dropbox. Can you also send me that stl file by email? My email is cmichel@sandia.gov.

That's weird that the RM3 example is not working. Can you try pulling the latest version of WEC-Sim, and add ``simu.paraview = 1``  to the input file of the RM3 example? Add this line anywhere after the simu variable has been initialized, and don't make any other changes. Once you have that working we can figure out why your specific example isn't working.

closed 2015-11-09 by cmichelen