WEC-Sim/WEC-Sim #52: How to model in WEC-SIM with 6DOF

ahagmuller, 2015-10-27:
Hi,
This question is related to Issue/Question #50.

We currently have 3 PTO blocks tied together for 3DOF in our model. This is, heave, pitch, surge, and sway. How can WEC-SIM be modified to analyze 6DOF, for heave, pitch, surge, sway, yaw, and roll.

Thanks in advance!

Alex

cmichelen, 2015-10-28:
There have been a lot of questions about this.mWEC-Sim outputs results in all 6DOFs. And yes, the default PTOs and Constraints in WEC-Sim only act in 3DOF (surge-heave-pitch). This does not mean that you cannot model 6DOF.

WEC-Sim is distributed as a Simulink library. For most cases, when building your model you only use blocks from the WEC-Sim library. But for more complex models (i.e. user defined forces, joints in different DOFs, etc.) you might have to use blocks from other libraries. These include blocks from the base Simulink libraries (math, etc.) as well as blocks from the SimMechanics library (which you have if you are using WEC-Sim). The SimMechanics Library is found under: ``Simscape > SimMechanics > Second Generation``.

For dealing with joints, SimMechanics has many options including several 6DOF joints. By combining WEC-Sim PTOs, Constraints, and SimMechanics joints, and coordinate transforms, you should be able to model any combination of DOFs you want. This does require some understanding of how SimMechanics coordinates and transforms work. I recommend taking a look at the SimMechanics User's Guide.

Another option for achieving different DOFs is to take a look at existing WEC-Sim PTOs/Constraints and modify them as desired.

Since there have been a lot of questions on this topic I will leave this issue open (but close #50) if anyone has more questions.

cmichelen, 2015-10-28:
Also, depending on your device and the wave conditions 3DOF might be an accurate assumption. For example, if you have a  symmetrical body in the direction of wave propagation your device should move in planar (3DOF) motion. If using 6DOF in such case you might get motion in the other DOFs but it would be numerical noise. In such a case you might be better off assuming no motions in those DOFs.

ahagmuller, 2015-10-28:
Hi Carlos ,

Thanks for the quick reply. Our device is axisymetrical, and that is why we were assuming 3dof would be acceptable for our application. 

We have modified the standard pto blocks to achieve 6 dof and are going to compare results from our 3dof analysis. 

I will try and post more on the specifics of our situation as time allows, someone may find our findings useful for a non-axisymetrical body. 

Regards,

Alex

cmichelen, 2015-10-28:
Thanks Alex, that would be great. 

Also, something that might be good to know when modeling in 6DOF is that you can apply mooring in any DOF through the ``body(i).mooring`` property. It takes 6x6 matrices for stiffness and damping, and a scalar for preTension. This mooring forces are applied at the cg. See #43

ahagmuller, 2015-10-28:
Hi Carlos,

Our solution so far has been to add a spherical joint at the body, and a
universal joint at the seafloor, with a "Z" pto connecting them,
approximating a tension connection between the two constraints.

The spherical joint at the body was done as a PTO, not a constraint, so we
can change the damping values and spring stiffness at that joint.

We have been using the mooring damping/stiffness matrix, but were wondering
if it still applies anything at the CG with this kind of configuration.

Our plan was to use the spherical joint damping and stiffness with a
transformation to approximate a catenary mooring line connection. For
example, the damping at the spherical joint is rotational damping, and the
mooring connection is 5m away from the CG. Since torque is angular
velocity*angular damping, for instance, we set this equal to the radial
distance*vertical mooring force, and solve for vertical mooring force to
estimate the catenary vertical mooring tension.

Does this sound reasonable, or should we stick with the mooring
damping/stiffness matrix?

Also, as you pointed out, the power and damping is not considerably
different from 3DOF, but the other 3DOF do show up as slight noise, but it
does appear to output correctly.

Thanks again,

Alex

ahagmuller, 2015-10-28:
Hi Carlos,
I have put the files for discussion below in our shared folder:

https://github.com/ahagmuller/WEC-SIM-AquaHarmonics_Shared/tree/master/AH_6DOF

This is using the RM3 float for conceptual purposes as well as the RM3.h5 file.

I have not finished the modifications for using the spherical constraint and universal constraint for outputs, but it should get the idea across for now.

Please let me know should you have any questions.

Best Regards,

Alex

cmichelen, 2015-11-02:
Hi Alex. 

Thanks for sharing that example! I wonder what the difference would be between using a spherical, universal, and gimbal joints. I think the dynamics might be the same, but the output DOFs might be different. I think these would be good to include in WEC-Sim. So if you do finish them feel free to do a pull request (and become a WEC-Sim contributor!).

cmichelen, 2015-11-02:
We just added a 6DOF floating constraint. If you pull the latest version of WEC-Sim you should have it. Both the 3DOF and 6DOF floating constraints can now have mooring (either through WEC-Sim or coupled to MoorDyn).

[image]

closed 2015-11-02 by cmichelen

ahagmuller, 2015-11-02:
Hi Carlos ,

I will try and get a drawing of the intended mechanics of what is modeled in that example. 

I think this could work well as a base for a lot of 6 DOF single body point absorbers with different PTO topologies.

For instance , with this example the universal joint is intended to act as a shackle to an anchor at the sea floor. The Z -PTO attached to that frame of reference has a line of action always between the universal joint and the spherical joint , which could be either a column type PTO like a rack and pinion or linear generator if constant damping in both directions is used.  The spherical joint allows free rotation in 3 DOF at the body connection. 

If one uses a user defined force with velocity sensing say in heave or surge or the magnitude of both of them to drive a switch between a damping force on positive velocity and zero force on negative velocity this can approximate a tension only connection. 

Also,
If you use a spherical joint at both the device end and the sea floor you will get a degenerate mass error. 

We are using rotational damping at the spherical end to help reduce numerical noise as well. 

When we get some time to button up the blocks I will send it your way. 

Thanks,

Alex

ahagmuller, 2015-11-02:
Hi Carlos ,

I haven't looked yet but is MoorDyn also open source? 

I had posted some questions regarding Mooring damping and stiffness and what I thought was the correct application of the matrix. 

I think it would be really helpful to the user to see a drawing of how the matrix is used base on the geometry of the device and how the equations are applied. I am still a little fuzzy on some I this myself.  

That being said, we are assuming that if we use damping on the spherical joint in the example we posted that this can be resolved to forces in x,y and z  based on the connection points of the morning lines. Then, the rotational damping*rotational velocity in that axis is equal to torque, and that torque divided by the effective torque arm length to the mooring connection will be the magnitude of the catenary mooring force which can be resolved into its components. Does this sound correct?

Thanks ,

Alex

cmichelen, 2015-11-02:
That would be great! Thanks.

I am moving your second question to a new thread.