WEC-Sim/WEC-Sim #59: Hydrostatic stiffness bug

cmichelen, 2015-11-04:
WEC-Sim currently assumes that the hydrostatic stiffness matrix has values only on one side of the diagonal. 

This is not true for BEMIO files created with AQWA. Not sure about WAMIT and NEMOH, but BEMIO should output the entire matrix.  

in bodyClass/hydroForcePre:
```matlab
k = obj.hydroData.hydro_coeffs.linear_restoring_stiffness;
obj.hydroForce.linearHydroRestCoef =  (k + k' - diag(diag(k))).*rho .*g;
```

should be: 
```matlab
k = obj.hydroData.hydro_coeffs.linear_restoring_stiffness;
obj.hydroForce.linearHydroRestCoef =  k.*rho .*g;
```
I will make this change after verifying what BEMIO does for WAMIT and NEMOH. @lawsonro3

ratanakso, 2015-11-05:
Carlos,

I was able to use pip for matplotlib, progressbar2 and astropy, but I can't
use pip with numpy scipy and h5py. Any suggestions?

Thanks,
Nak

cmichelen, 2015-11-05:
Checked: WAMIT is the only one that outputs half the matrix. For NEMOH and AQWA BEMIO outputs the whole matrix.

Changed BEMIO wamit.py as follows (added lines 165-166,172-173):
[image]

Will push BEMIO and bodyClass changes soon. Need to create new h5 files for all the tutorials/verification cases.

closed 2015-11-05 by cmichelen