WEC-Sim/WEC-Sim #60: Trouble installing bemio

MalteTwo, 2015-11-06:
Hi folks,
i´m desperately trying to find a possibility to use WEC-Sim. As I figured, I need Nemoh to get hydrodynamic coefficients, which I then have to trynsform using Bemio, so I can use them as input for WEC-Sim. I managed to set up Nemoh, but now I´m struggling.

Trying to install bemio in command prompt, I get the following error-message:

Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from bemio.__version__ import base
  File "C:\...\bemio-master\bemio\__init__.py", line 1, in <module>
    from __version__ import base, full
ImportError: No module named '__version__'

I exactly followed the following the instructions given on http://wec-sim.github.io/bemio/installing.html#downloading-bemio. Does anybody have an idea, why it doesn´t work? I didnt´change anything in the bemio-folders I´ve downloaded, so why can python not find the '__version__' file? its in the bemio folder!

I´m really not a computer expert, so I hope someone might be able to help me learn about all this - its exciting, but also frustrating at times. Any help is highly appreciated - thanks a lot! Regards
Malte

ratanakso, 2015-11-10:
Hi Malte,

I recently had a chance to play with Bemio and I thought I could share what I did.
1. Install PythonXY (Windows). When it is completed, you should have numpy, scipy, matplotlib,astropy, and h5py. The only missing package is progressbar2. 
2. Open your cmd.exe and type "pip"
3. If you have pip, then you can do this "pip install progressbar2" By now you should have everything you needed. Close your cmd.exe
4. Use your Git Shell to clone WEC-Sim/bemio by typing "git clone git@github.com:WEC-Sim/bemio.git"
5. Open your cmd.exe again and cd to bemio directory
6. Type "python setup.py install --user"
7. Type "python install_test.py"

Good luck!
Nak

cmichelen, 2015-11-10:
Hi Malte,

I would recommend reinstalling python and bemio from scratch. Python XY is a good option (Nak explained the process for installing it.) If you do reinstall Python be sure to uninstall/delete your current python.

Also, this user had a similar issue (in the bemio issues page): https://github.com/WEC-Sim/bemio/issues/18 
I have asked him if he was able to solve it.

lawsonro3, 2015-11-10:
Hi Malte,

Sorry you are still having trouble. I just checked everything with a clean installation of Anaconda and a freshly downloaded copy of bemio on a Windows 7 system. Please make sure you have the latest version of bemio. The code blocks below shows the steps I used to successfully install and test bemio.

If the steps below don't work, please reinstall everything and try again.

Hope that helps,

-Mike

## Check python version and folder contents
```shell
Windows PowerShell
Copyright (C) 2009 Microsoft Corporation. All rights reserved.

C:\Users\mlawson\Documents\GitHub\bemio [master]> python --version
Python 2.7.10 :: Anaconda 2.3.0 (64-bit)
C:\Users\mlawson\Documents\GitHub\bemio [master]> ls

    Directory: C:\Users\mlawson\Documents\GitHub\bemio

Mode                LastWriteTime     Length Name
----                -------------     ------ ----
d----         8/17/2015   2:04 PM            bemio
d----         8/17/2015   2:05 PM            bemio.egg-info
d----         8/17/2015   2:05 PM            build
d----         8/17/2015   2:05 PM            dist
d----        11/10/2015  11:37 AM            doc
d----         8/17/2015   1:45 PM            tutorials
-a---         7/29/2015   4:07 PM        862 .gitignore
-a---         8/17/2015   1:45 PM        203 clean_install.py
-a---        11/10/2015  11:36 AM       1697 install_test.py
-a---         7/29/2015   4:07 PM        325 README.md
-a---         5/12/2015  10:06 AM         26 setup.cfg
-a---         8/17/2015   1:45 PM        965 setup.py
-a---         7/29/2015   4:07 PM        133 USERS_GUIDE.html
```
## Install bemio
```shell
C:\Users\mlawson\Documents\GitHub\bemio [master]> python setup.py install --user

running install
running bdist_egg
running egg_info
writing requirements to bemio.egg-info\requires.txt
writing bemio.egg-info\PKG-INFO
writing top-level names to bemio.egg-info\top_level.txt
writing dependency_links to bemio.egg-info\dependency_links.txt
reading manifest file 'bemio.egg-info\SOURCES.txt'
writing manifest file 'bemio.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build\bdist.win-amd64\egg
creating build\bdist.win-amd64\egg\bemio
creating build\bdist.win-amd64\egg\bemio\data_structures
copying build\lib\bemio\data_structures\bem.py -> build\bdist.win-amd64\egg\bemi
o\data_structures
copying build\lib\bemio\data_structures\wave_excitation.py -> build\bdist.win-am
d64\egg\bemio\data_structures
copying build\lib\bemio\data_structures\__init__.py -> build\bdist.win-amd64\egg
\bemio\data_structures
creating build\bdist.win-amd64\egg\bemio\io
copying build\lib\bemio\io\aqwa.py -> build\bdist.win-amd64\egg\bemio\io
copying build\lib\bemio\io\nemoh.py -> build\bdist.win-amd64\egg\bemio\io
copying build\lib\bemio\io\output.py -> build\bdist.win-amd64\egg\bemio\io
copying build\lib\bemio\io\wamit.py -> build\bdist.win-amd64\egg\bemio\io
copying build\lib\bemio\io\__init__.py -> build\bdist.win-amd64\egg\bemio\io
creating build\bdist.win-amd64\egg\bemio\mesh_utilities
copying build\lib\bemio\mesh_utilities\mesh.py -> build\bdist.win-amd64\egg\bemi
[... 238 lines cut ...]
body3:
   * Setting added mass at infinite frequency to added mass at omega = 20.943952
6
   * Setting added mass at zero frequency to added mass at omega = 0.7222053
   * Calculating real and imaginary excitation  components.
        Un-scaling hydro coefficients for body body3 by rho, g, and w...
Radiation damping IRF for body1:100%|#########################################|
Excitation force IRF for body1:100%|##########################################|
Radiation damping IRF for body2:100%|#########################################|
Excitation force IRF for body2:100%|##########################################|
Radiation damping IRF for body3:100%|#########################################|
Excitation force IRF for body3:100%|##########################################|
Writing HDF5 data to C:\Users\mlawson\Documents\GitHub\bemio\tutorials\aqwa\data
\aqwa_example_data.h5
        Radiation state space coefficients for body1 were not written.
        Radiation state space coefficients for body2 were not written.
        Radiation state space coefficients for body3 were not written.
****The tutorials/aqwa test case ran successfully****

Test cases runs completed:
        tutorials/wamit/COER_hydrodynamic_modeling_comp: SUCCESS
        tutorials/wamit/wec3: SUCCESS
        tutorials/nemoh: SUCCESS
        tutorials/aqwa: SUCCESS
```

MalteTwo, 2015-12-10:
Dear Mike,
Excuse my long abscence, I´ve been busy with some other stuff lately,
but I´d still like to get back to you on the bemio topic -
unfortunately, it still does not work for me. I hope its ok for you if I
bother you with this stuff again - I really want to know what is wrong
with what I do! I had hoped that the problem was my Windows XP
environment, so I dropped my old Laptop and got a fresh, Windows 7
(32bit system) one. Here´s exactly what I did on that one: 
1) I installed GitHub Desktop
2) I installed Anaconda from https://www.continuum.io/downloads#_windows
(version Python 3.5 for 32 bit)
3) I opened the "Git Shell" and typed git clone
https://github.com/WEC-Sim/bemio.git to download the latest bemio
version

So far, so good. I assume all this has worked properly, for this is what
I get following your example (see your code below; I marked red what I
copied from Git Shell): 

C:\Users\Malte\Documents\GitHub\bemio [master]> python --version
Python 3.5.0 :: Anaconda 2.4.0 (32-bit)

--> Python is properly (?) installed

C:\Users\Malte\Documents\GitHub\bemio [master]> dir

    Verzeichnis: C:\Users\Malte\Documents\GitHub\bemio

Mode                LastWriteTime     Length Name
----                -------------     ------ ----
d----        07.12.2015     20:59            bemio
d----        07.12.2015     20:21            doc
d----        07.12.2015     20:21            tutorials
-a---        07.12.2015     20:21        862 .gitignore
-a---        07.12.2015     20:21        203 clean_install.py
-a---        07.12.2015     20:21       1697 install_test.py
-a---        07.12.2015     20:21        325 README.md
-a---        07.12.2015     20:21         26 setup.cfg
-a---        07.12.2015     20:21        965 setup.py
-a---        07.12.2015     20:21        133 USERS_GUIDE.html

--> also Bemio is downloaded; I noticed that in contrast to your version
(see below) I do not have the files "bemio.egg-info", "build", and
"dist". Might this be a problem? I can´t see them on bemio´s GitHub
repository (https://github.com/WEC-Sim/bemio/) though.

4) I added python to my system
5) I navigate to the bemio-folder in the Git Shell and enter python
setup.py install --user; see what I get:  

C:\Users\Malte\Documents\GitHub\bemio [master]> python setup.py install
--user
Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from bemio.__version__ import base
  File "C:\Users\Malte\Documents\GitHub\bemio\bemio\__init__.py", line
1, in <mo
dule>
    from __version__ import base, full
ImportError: No module named '__version__'

I also tried the same command (python setup.py install --user) in my
normal windows command line, but it gives the same output. 

As you see, I EXACTLY follow your instructions, but I have the same
problem on windows xp, windows 7, and no matter how often I tried to
re-install everything. Its fantastic:) Do you have any idea, what´s
going wrong? 
Once again, I appologise for keeping you busy with this trivial issue,
but I see no way out myself. Thanks a million!!
[... 380 lines cut ...]
        Un-scaling hydro coefficients for body body3 by rho, g, and w...
Radiation damping IRF for
body1:100%|#########################################|
Excitation force IRF for
body1:100%|##########################################|
Radiation damping IRF for
body2:100%|#########################################|
Excitation force IRF for
body2:100%|##########################################|
Radiation damping IRF for
body3:100%|#########################################|
Excitation force IRF for
body3:100%|##########################################|
Writing HDF5 data to
C:\Users\mlawson\Documents\GitHub\bemio\tutorials\aqwa\data
\aqwa_example_data.h5
        Radiation state space coefficients for body1 were not written.
        Radiation state space coefficients for body2 were not written.
        Radiation state space coefficients for body3 were not written.
****The tutorials/aqwa test case ran successfully****
Test cases runs completed:
        tutorials/wamit/COER_hydrodynamic_modeling_comp: SUCCESS
        tutorials/wamit/wec3: SUCCESS
        tutorials/nemoh: SUCCESS
        tutorials/aqwa: SUCCESS

kmruehl, 2016-01-28:
@MalteTwo has this issue been resolved?

closed 2016-02-09 by cmichelen