WEC-Sim/WEC-Sim #61: domain size/ up-wave and down-wave distance question

ahagmuller, 2015-11-07:
Hi,
I am trying to find information for WEC-Sim regarding the domain size, Up-Wave and Down-Wave distance. Where is this found?
Thanks,
Alex

cmichelen, 2015-11-09:
Hi Alex,

``simu.domainsize`` is used for visualization only. and it is used inside the reference frame block. 

``waves.waveAmpTime`` has the wave elevation time-history at the (0,0,0) location.

closed 2015-11-12 by cmichelen

ahagmuller, 2015-11-12:
Hi Carlos,
I was also looking into the BEM side of things, is there anything there that can be changed? I couldn't find anything in the NEMOH documentation that would do this except for visualization purposes. Do you know if WAMIT or AQWA have the ability to change this?
Thanks,
Alex

yuyihsiang, 2015-11-12:
Hi Alex

In those BEM models, based on the numerical formulation, the outer boundary was assumed at infinity. Therefore, you only need to panelize the body. The only except is when you are using the second-order wave theory model in WAMIT. However, we only use linear wave theory in WAMIT, AQWA and NEMOH to calculate those hydrodynamic coefficients. Therefore, the domain size is only for visualization purposes.

Thanks, 

--Yi-Hsiang

ahagmuller, 2015-11-12:
Got it, thanks Yi.

Alex