WEC-Sim/WEC-Sim #62: STL normal vector calculation issue for certain files.

cmichelen, 2015-11-10:
In general, the order of points in an stl should be given such that the normal of each triangle is obtained through the right hand rule. Our calculations assume this. So does the SimMechanics visualization. But some software (e.g. SolidWorks when using shading) don't give the points in ht right order. The STL does provide the normal vectors.

This is a problem when using non-linear hydro with a an stl that doesn't follow the right hand rule.

Suggested action: 
Use the normal vectors from the STL, also calculate them using the right hand rule. If they disagree, give a warning.

From https://en.wikipedia.org/wiki/STL_%28file_format%29#The_facet_normal

```
In both ASCII and binary versions of STL, the facet normal should be a unit vector pointing outwards from the solid object. In most software this may be set to (0,0,0), and the software will automatically calculate a normal based on the order of the triangle vertices using the "right-hand rule". Some STL loaders (e.g. the STL plugin for Art of Illusion) check that the normal in the file agrees with the normal they calculate using the right-hand rule and warn the user when it does not. Other software may ignore the facet normal entirely and use only the right-hand rule. Although it is rare to specify a normal that cannot be calculated using the right-hand rule, in order to be entirely portable, a file should both provide the facet normal and order the vertices appropriately. A notable exception is SolidWorks, which uses the normal for shading effects.
```

bradling, 2015-12-02:
I have found that the code that checks for this problem will sometimes output a warning that the STL's are not consistent with the right hand rule when some of the elements normal vectors are aligned with one of the global axes. 

After seeing this error for a few different geometries I was simulating, I plotted the normals as recorded in the STL, and the normals calculated using the cross product. Both sets of normals pointed outward. PS - I am using Salome-Meca to generate my STL files.

I think this is a numerical rounding error, where a small error on opposite sides of zero changes the sign of one component of the normal vector. I don't know if its worth modifying the code to be more numerically stable, but it is an issue I have run into.

cmichelen, 2016-01-07:
I commented out this warning because I cannot come up with an easy and robust way of checking this. The current implementation should always work (because the normal vectors reported by the STL should be correct regardless of the software used to create it). The only problem is that if the order of vertices is incorrect, Matlab won't display the geometry correctly in the SimMechanics Explorer and the user might not know why.

closed 2016-01-07 by cmichelen