WEC-Sim/WEC-Sim #63: Remove large binary files from repo

petebachant, 2015-11-10:
A zip of the current master branch is 300+ MB. Notice that, for example, in https://github.com/WEC-Sim/WEC-Sim/commit/eeaf882834ac79731d9cf715a78ee3a4dbd54301, changes are being committed to large HDF5 files, which is probably increasing the size of the repo dramatically, since Git cannot version binary files (it just saves _every_ version). You may want to put the large binary files from the tutorials in a [figshare](http://figshare.com) repo and have the newest versions downloaded automatically, or look into using [Git large file storage](https://git-lfs.github.com/) to make cloning the repository faster and easier for users.

cmichelen, 2015-11-11:
We should look into that. One option for most users, who do not need to checkout older versions, is to use depth=1 when cloning.

```
git clone --depth 1 git@github.com:WEC-Sim/WEC-Sim.git
```

bradling, 2015-11-12:
I wonder how much benefit hosting the HDF5 file externally would have, because the simulink models slx files are also binary, and those change often. Don't you wish github could diff simulink files? It would make life much easier

petebachant, 2015-11-12:
Depends on how often the *.slx files change really. At least they appear to be small. You could ask MathWorks to start saving them as plain text (good luck), or develop a plugin for diffing/merging them with Git (even more good luck).

It would be possible to use Git Large File Storage for both the *.slx and HDF5 files, which is compatible with GitHub: https://git-lfs.github.com/. You'd probably need to do a lot of rebasing to slim down the repo though. It's a tough situation, but it might be worth it to make the download 5 times smaller.

closed 2017-05-09 by kmruehl