WEC-Sim/WEC-Sim #72: RM3MoorDyn Bug

lianggege, 2015-12-01:
By running case RM3MoorDyn, I find the following bugs:
Error using wecSim (line 210)
Error evaluating 'InitFcn' callback of SubSystem block 'RM3MoorDyn/Floating (3DOF)/Mooring/MoorDyn/MoorDyn'.

Caused by:
    Error using wecSim (line 210)
    Could not find file MoorDyn.h. 

Does anybody has similar experience?

yuyihsiang, 2015-12-02:
It have test it, and it works fine on my Mac and a Windows machine. 

1. Do you have the MoorDyn.h file in the "WEC-Sim Path"/source/functions/moorDyn/ folder?
2, What OS are you using? Currently, the WEC-Sim/MoorDyn coupling only works for Windows and OSX but not for Linux 

Thanks,

--Yi-Hsiang

lianggege, 2015-12-16:
Yes, I have the file ‘MoorDyn.h' in the path.
[image]
I downloaded the 'WEC-SIM-master.zip' from the forum, and didn't make any changes of the files.

cmichelen, 2015-12-23:
Make sure that when running WEC-Sim you ``add all folders and subfolders`` to the MATLAB path. I don't know what else it could be ...

closed 2016-02-09 by cmichelen