WEC-Sim/WEC-Sim #75: PTO and constraints

evemoralesc, 2015-12-08:
Hi everybody,

I'm modeling an OSWEC and I have a question about the PTO global location.

My flap must rotate about the base and I want to locate a piston in the middle of the flap, but when I declare the PTO location (pto(1).loc) the flap rotates about this point. 

Then, I'm wondering where I must declare the piston location. Should I set up a pitch constraint plus the PTO mechanism?

Thanks in advance for your answer

cmichelen, 2015-12-09:
Whether you use a Pitch Constraint or a Pitch PTO the location must be the point of rotation. For the piston, a simple schematic of your setup would be helpful. Is it something like this (http://wec-sim.github.io/WEC-Sim/tutorials.html#oswec-with-pto-sim)?

cmichelen, 2015-12-18:
I'm closing this. Re-open if you still need help.

closed 2015-12-18 by cmichelen