WEC-Sim/WEC-Sim #76: Morison Element water particle velocity z component

bradling, 2015-12-12:
Hello,

I think the z-component of the water particle velocity used in the morison element calculation has the incorrect sign. I am going to do some more investigation into this over the weekend and see if I can demonstrate this with a simple model.

bradling, 2015-12-14:
I did some more research into this, and I think the issue is not the sign of the z-component of fluid velocity, but rather the sign of the entire morison force term.

I tested this by creating a simple 1 body model, with a prescribed motion heave only PTO. I then included a morison drag term at the CG of the body, and simulated with no input waves. This should be equivalent to using a viscous drag term, as the orbital fluid velocity is zero. However, the drag force has the opposite sign from a model using a viscous drag term, see plot below. If needed I can upload the model and input file I used to generate these results.

[image]

bradling, 2015-12-18:
Attached is my test case. The h5 file is a dummy file with zeros for hydro force coefficients.

[morison_test_case.zip](https://github.com/WEC-Sim/WEC-Sim/files/66872/morison_test_case.zip)

cmichelen, 2015-12-18:
Hi Brad,

I haven't had a chance to look at the Morrison Element implementation in detail, but I played with the test case you shared and switching the sign in the addition block seems to solve the problem, like you suggest. The best way to solve it would have been to go into the ME blocks and have it output force in the correct direction, but for now this should be good enough.  I made this change and pushed. 

This are the results before the change:
[image]
[image]

And this is after:
[image]
[image]

cmichelen, 2015-12-18:
Let us know if you find any other bugs with the morison elements. We just implemented it very recently.

closed 2015-12-18 by cmichelen