WEC-Sim/WEC-Sim #77: Theory documentation

krisRaj, 2015-12-14:
Hi,

Let me point out a difference in the theory documentation. Perhaps this is a typo.

In  http://wec-sim.github.io/WEC-Sim/theory.html,  under Convolution Integral Formulation the formula for Fext is given one way ( Looks like the integral should be way out) and in http://prod-http-80-800498448.us-east-1.elb.amazonaws.com/w/images/c/c7/WEC_Sim_User_Manual_v1.0.pdf, Eq. 2.5, the formula for Fext is given in another way ( looks like dwr should be inside the sqrt sign). How is it coded in wecSim?

Thank you

Kumar

cmichelen, 2015-12-23:
Thanks for pointing this out! We no longer maintain that PDF documentation, instead we are using the website  http://wec-sim.github.io/WEC-Sim/. 

It seems to be wrong in both places... All terms except for Rf should be inside the integral (like in the pdf). Rf can be taken out because it is not a function of wr. But the dwr term should also be under the square root (like in the website). I think it should be this:

[image]

In WEC-Sim this implemented as a sumation over i=1:N where N is the number of frequencies in the frequency vector. 
```
F_wave = sum( F_wave(i))
F_wave(i) =  cos(w(i) * t + phi(i))  *  Re{Fext(i)}  *  sqrt(S(i) * dw)    
                   - sin(w(i) * t + phi(i))  *  Im{Fext(i)}  *  sqrt(S(i) * dw)
```

Under ``WECSim_Lib/Body Elements/Rigid Body/Hydrodynamic Body/Wave Diffraction and Excitation Force Calculation/Linear Wave Excitation Force Variant Subsystem/Irregular Wave  Excitation Force``:

[image]

[image]

I will look at this again after the holidays and change the documentation if this is right.

krisRaj, 2016-01-05:
That looks good. Thank you Carlos,

cmichelen, 2016-01-06:
This is now fixed in the documentation: http://wec-sim.github.io/WEC-Sim/theory.html#convolution-integral-formulation

closed 2016-01-06 by cmichelen