WEC-Sim/WEC-Sim #78: Hydrodynamic interaction within Rigid Body block

lianggege, 2015-12-20:
Hi everybody,
By looking inside the Rigid Body block (Hydrodynamic body), I find that the total external force (F_Total) that the body suffers is only depended on its own displacement, velocity and acceleration. According to my knowledge, in a two floater condition, for example the RM3 case, the radiation force for floater A not only depends on its own motion, but also on the motion of floater B. So, could I ask how WEC-Sim deals with hydrodynamic interaction issue?

cmichelen, 2015-12-23:
Usually the forces due to the body's own motion are much larger than those due to the motions of the other bodies and by default WEC-Sim runs without body-to-body interactions.

You can turn body-to-body interactions ON by setting ``simu.b2b = 1;`` in your input file. If you look inside the ``Rigid Body`` block you will find a block called ``B2B Subsystem``. From the screenshot below you can see that the velocity used for the Radiation Force (added mass and radiation damping) come from the B2B Subsystem block.

[image]

In the case of four bodies the B2B subsystem looks like this:

[image]

The green ``From`` blocks correspond to the red ``GoTo`` in each body (you can see them for body(1) in the first screenshot).

lianggege, 2015-12-23:
got it， thanks a lot！

closed 2016-01-06 by cmichelen