WEC-Sim/WEC-Sim #83: Paraview visualization bugs

Toby-D, 2015-12-30:
Carlos, great job with the Paraview visualization capabilities! Just being able to see the waves is a huge improvement over the SimExplorer view.

A couple small bugs I came across while I was getting it working:

 1. Default Nemoh body names that include file separators cause problems (e.g. in WEC-Sim line 263)

A simple workaround is to rename the bodies in HDFview. A more robust solution might strip the directory from bodyname.

 2. If there's an error writing the paraview files, they stay open and can't be modified until MATLAB is restarted.

I added a quick and dirty "fclose all" at the beginning of wecSim.m that fixed this. I believe that a better solution would be to incorporate an fclose into a try-catch when opening files for writing.

 3. When writing the paraview files, MATLAB confuses the fileseps in the file names as the start of control characters (responseClass.m lines 136 and 143).

Replacing "filesep" with "filesep filesep" fixes this, at least on my system*. But I don't know whether that's robust across different platforms. Maybe it would be better to create the string in a separate line and then reference it in the fprintf statement.

*MATLAB R2015b (Yes, I know it's not supported...), on Windows 10. WEC-Sim synced 12/30/2015 (f997b67).

Thanks again for all the work you've put into to this!

bradling, 2015-12-30:
Hi Toby,

I have run into similar issues using Paraview. I use wecSim on a windows 7 machine, and I often have file separators in my body names (I use NEMOH to calculate hydro coefficients). I actually made a few modifications to responseClass and bodyClass that have fixed these issues for me at least. 

I think what I did is two things: one, for writing paraview files i replace all filesep's in the body name with an "_". And for the third issue, I check if filesep == "/", if so I replace it with "//" so it prints properly.

@cmichelen  - I would be happy to share these changes if you are interested. No need for you to duplicate things I have already done. But as I said, I have only tested these changes on windows 7 - I don't have a mac or linux system to test on.

cmichelen, 2016-01-06:
I will look into this. I have had trouble with Matlab confusing 'filesep' with a start of a control character in some systems, but maybe 'filesep filesep' would work as you suggest.

cmichelen, 2016-01-06:
1. I changed BEMIO to strip the directory and file extension from the body name. If you pull the latest BEMIO and create the h5 file again the bodyname should not have any file separators

2. The problem is that to use `fclose` inside a `try` I need to know the file id for the file I am trying to close. Can you think of a way of doing this? What I would do is just type `fclose all;` into matlab after WEC-Sim fails, instead of modifying WEC-Sim to do this everytime. 

3. @bradling it would be great if you could share your solution to the third issue: using `filesep filsep`. I run WEC-Sim on a Mac, so I can check if it works for me, and then accept the pull request. Thanks!

cmichelen, 2016-01-07:
Thanks @bradling !

closed 2016-01-07 by cmichelen