WEC-Sim/WEC-Sim #84: non linear wecSim + irregular waves

krisRaj, 2016-01-05:
Hi All,

I followed Yi-Hsiang's post (https://github.com/WEC-Sim/WEC-Sim/issues/34) and downloaded the latest version of wecSim today. But when I run wecSim nonlinear + irregular waves , whether  PM spectrum, or user specified I get the following error:
///////////////////Error 
Error using wecSim (line 211)
Derivative of state 'h_and_f_rev1.Floating.Planar_Joint.Px.v' in block 'h_and_f_rev1/Floating/Planar
Joint' at time 0.0 is not finite. The simulation will be stopped. There may be a singularity in the
solution.  If not, try reducing the step size (either by reducing the fixed step size or by
tightening the error tolerances)
//////////////////
The time step is already very small. When I switch off the non linear part, wecSim runs without any problem. 

Thanks in advance
Kumar

krisRaj, 2016-01-05:
Let me give a little more info. The wec is similar to an OSWEC , but with the base also floating. see for example this thread https://github.com/WEC-Sim/WEC-Sim/issues/55

The error seems to be coming from the constraint(1)Floating. I am a little surprised because this component has never given trouble before

Thanks again
Kumar

yuyihsiang, 2016-01-05:
Hi Kumar

If you use PM for the nonlinear option, it should work. I have just tasted on my machine using the RM3 tutorial, and it worked fine. I also noticed that the error message you got does not seem to be an error directly due to nonlinear option. I can take a look on your model  to figure out what is going on if you can send it to me.

Also note that the nonlinear option does not work for user specified wave train. The nonlinear option requires to calculate the wave induced pressure at each panel on the wetted part of the body surface. Therefore, user specified wave train would not work because it was given at the mean water surface position with x and y equal to zero. There is a way to get around that is to use do a FFT to obtain the spectrum and the corresponding phase for the wave. Then use "User-Defined Spectrum” to read in the spectrum. However, the current version does not allow the model to read in the phase. We are currently working on that and should be able to have that option in the model soon.
Thanks,

--Yi-Hsiang

cmichelen, 2016-01-06:
Are you using ramp time? I have had similar issues when setting ``rampT=0``.

yuyihsiang, 2016-01-06:
Hi Kumar

Good call Carlos

That's THE PROBLEM. For now, I believe you can set simu.rampT="small value", and it should work. I would recommend you set simu.rampT=10 x peak period to avoid the transient. I'll check the case you sent me later this morning to make sure it works.

Cheers,

Yi-Hsiang

krisRaj, 2016-01-06:
Hi Yi-Hsiang, Carlos,

Thank you both for the suggestions. I tried changing simu.rampT =10*peakperiod (60 seconds). But the problem persists. What about the simu.CITime. Currently it is set to 20. and this is the .py file showing details of how the  .h5 was generated.
/////////////////////////////////////////////////////////////////////
from bemio.io.wamit import read
from bemio.io.output import write_hdf5

# Load the data using the wamit module.
wamit_data = read(out_file='test05.out')

# Calculate IRF and plot using the wamit module
for i in xrange(wamit_data.body[0].num_bodies): #wamit_data.body[0].num_bodies
 	wamit_data.body[i].calc_irf_radiation(t_end=20, n_t=501, n_w=501, w_max=10.)
	wamit_data.body[i].calc_ss_radiation(max_order=5, r2_thresh=0.90)
	wamit_data.body[i].calc_irf_excitation(t_end=30. , n_t=501, n_w=501)

# Save the data in the hdf5 format.
write_hdf5(wamit_data)
////////////////////////////////////////////////////////////////

Thanks again
Kumar

yuyihsiang, 2016-01-07:
Hi Kumar

I have updated the WEC-Sim lib to fix the issue you have. The problem is that we are using intermediate depth formula when the water depth is not infinite even it is in the deep water region. However, Cosh (kh) function becomes so large if you have a very large k*h (k is wavenumber and h is water depth). Mathematically, a very large value divid by another large value can give you a reasonable solution. However, MATLAB can not handle that. Your case unfortunately made it worse because of the high frequency data you have, which results in a large wavenumber.

I modify the WEC-Sim model so that it use deep water theory when water depth > 0.5 wave length. I have tested it using your model, and it seems working. Let me know if it solves your problem

Thanks,
 
Yi-Hsiang

krisRaj, 2016-01-08:
Hi Yi-Hsiang,

THANK YOU VERY MUCH FOR THE MODS. I have the case WORKING now. I checked with a couple of PM spectra varying the periods with a generic PTO . They are all working fine.  Looks  like your fix has solved the issue. 

I need a small clarification. I see the simu.b2b =1 setting. So has the body2body interaction fully resolved in the wecSim now?. I believe this was an issue in the EWTEC 2015 paper.

Cheers

Kumar

cmichelen, 2016-02-09:
Hi Kumar,

Yes, body-to-body radiation forces are fully implemented now and can be turned on with the 'simu.b2b' flag.

closed 2016-02-09 by cmichelen