WEC-Sim/WEC-Sim #89: old version simulate

Owen0708, 2016-01-20:
Hi,
I am still use the old version WECSim, because install the new version is really difficult for me. 
and here is the error shown in below when I am use the floating constrant block to calculate the three DOF motion that have been bothering me for long time.
Derivative of state 'Floating.Planar_Joint.Px.v' in block 'test/Floating/Planar Joint' at time 7.65 is not finite.
The simulation will be stopped. There may be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances)
[image]
I think the problem is in my input "wamit.out", because I am use the NEMOH output transform to wamit.out format.
Is the input wave period has some range or limit?
and when the calculate stop, my input data are totally different in my MATLAB Workspace, like body mass & momOfInertia.
but if I change a bigger parameter "Volumes (VOLX,VOLY,VOLZ)" in the wamit.out, the error will not happen, but it means the different float in my simulate.

Can anyone help me? I would appreciate it if you could help me.

cmichelen, 2016-01-21:
Hi,

A few comments: 

* I highly recommend getting the newest version of WEC-Sim. WEC-Sim is a relatively new code and we are currently fixing bugs and adding new features.
* If you get the newest version you can use your NEMOH outputs directly, without converting to WAMIT.
* WEC-Sim changes Mass, Moment of Inertia, and Added Mass during simulation. It then cahnges it back in the workspace when the simulation is done. BUT if the simulation crashes the workspace will have the 'internal' mass properties. Basically this is not an issue, you don't have to worry about it.
*Try the new version, and try a smaller time-step. If you still get this issue, we can look into it more.

Hope this helps!

Owen0708, 2016-01-22:
Hi,
I still trying to install the new verion WEC-Sim!
Thank you for your kind regard.

Owen0708, 2016-01-30:
Hi,
I use another computer to install the BEMIO, It's can run success.
then use the : python install_test.py to transform my NEMOH data to the NEMOH.h5.(just change the data in the tutorials\nemoh\data\one_body.)
but still have a question, the problem as the below graph, like this:
[image]
also I found that some people with same issue, and they post the question in the bemio forun.  
so I am try to change my CIC time, could anybody kindly help me, how to select a t_end to a number where the IRD decays appropriately to 0?
I am confuse in there relationship between CIC time & IRF.
and here is my input in Nemoh.cal & Bemio run.py.
Nemoh.cal: 
[image]
run.py:
[image]
another question is : if I set the wave frequency in a range between 1(rad/s) and 10(rad/s) in the Nemoh program, the WECSim input file "wecSimInputFile" may I set a wave period exceed the range from the Nemoh (ex. the wave period set 12.566 (s) = wave frequency 0.5 (rad/s) )?

Thank you
owen

cmichelen, 2016-02-09:
When you create your h5 file (using a python script) you can play with the inputs to the `calc_irf_radiation` function. You can then see and plot the results either through H5View or in Matlab after running WEC-Sim. They would be in `body(1).hydroData.hydro_coeffs.radiation_damping.impulse_response_fun`

For example, you could plot:
```Matlab
plot(body(1).hydroData.hydro_coeffs.radiation_damping.impulse_response_fun.t, squeeze(body(1).hydroData.hydro_coeffs.radiation_damping.impulse_response_fun.K(3,3,:)))
```
You can then control your  simu.CITime in your input file. The CITime can be anything less than or equal to the length of the irf from your h5 file. You are just choosing were to cut it off.

No, you cannot set a wave frequency outside you BEM range.

Hope this helps,

Carlos

closed 2016-02-17 by cmichelen