WEC-Sim/WEC-Sim #90: Variable Step Solvers

ImpulseResponseFunction, 2016-01-29:
ODE4 seems to be WEC-Sim's default solver; can one readily use a variable step solver with WEC-Sim? 

Does anyone know, offhand, if a variable step solver would cause WEC-Sim's convolution integral to give erroneous results. (Yeah, yeah, I know about WEC-Sim's state-space approximation ability for fluid memory effects)

cmichelen, 2016-02-02:
WEC-Sim should run with variable time step, but as you point out the convolution calculation would be erroneous. I believe that is the only part of the code that requires constant time-step. So you should be able to use variable time-steps for 'regular' and 'noWave' cases, or with any other case using the state-space implementation. I have never tried any of this, but you could check that this is the case by running the same simulation with constant and variable time-steps and comparing the radiation damping forces.

If you look into this more, feel free to share your findings here, I am sure other people would be interested in running variable time-step.

kmruehl, 2016-02-03:
I second Carlos' assessment.  The only part of the WEC-Sim code that requires a fixed time-step is the convolution integral, so if you run with code without 'CIC' you should be able to run variable time-step without issue. State Space, 'regular', 'noWave' should all work, but 'regularCIC' and 'noWaveCIC' will have errors.

closed 2016-02-09 by cmichelen