WEC-Sim/WEC-Sim #96: Rotation Matrix Error

ahagmuller, 2016-03-03:
Hello,
We just went through and updated to the latest WEC-Sim version as well as bemio, and revised our .h5 file as well. We started with a new .slx file and are trying to simulate a simple 2D version of our device in the new version. Starting with just a fixed constraint and a Heave PTO, the simulation worked, then we added one rotation constraint in place of the fixed constraint, and that worked. Finally we added another rotation constraint at the device, and we got the following error:

Error using wecSim (line 217)
Error evaluating parameter 'RotationMatrix' in 'AH_TG2_1/Translational PTO/Rigid  Transform'

Caused by:
    Error using wecSim (line 217)
    Expected one output from a curly brace or dot indexing expression, but there were 2 results.
 
Also, when we tried to just have the heave PTO directly connected from the device to the seafloor, with no constraint we got this error which we did not get in the previous version:

Error using wecSim (line 217)
Error due to multiple causes.

Caused by:
    Error using wecSim (line 217)
    Error evaluating parameter 'TranslationCartesianOffset' in 'AH_TG2_1/Translational PTO/Rigid  Transform'
        Error using wecSim (line 217)
        Undefined variable "constraint" or class "constraint.loc".
    Error using wecSim (line 217)
    Error evaluating parameter 'RotationMatrix' in 'AH_TG2_1/Translational PTO/Rigid  Transform'
        Error using wecSim (line 217)
        Undefined variable "constraint" or class "constraint.orientation.rotationMatrix".

Here is a screenshot of the .slx file:

[image]

Any help is greatly appreciated!

Regards,
Alex

cmichelen, 2016-03-03:
This was a bug on the PTO blocks in the library. If you pull again it should be fixed. Let me know if that is not the case.

closed 2016-03-03 by cmichelen

ahagmuller, 2016-03-03:
Will do thanks Carlos .

ahagmuller, 2016-03-04:
Just an update,

Tried with two rotational constraints as well as no constraint and no errors after latest pull.
Thanks again,

Alex