WEC-Sim/WEC-Sim #97: Paraview Macro Error

ahagmuller, 2016-03-06:
Hi,

When we run the paraview macros pressureGlyphs.py and WEC-Sim.py, we get the following errors:

Traceback (most recent call last):
  File "C:\Users\Pirate\Documents\GitHub\WEC-Sim\source\paraview_macros\pressureGlyphs.py", line 2, in <module>
    from paraview.simple import *
ImportError: No module named paraview.simple
>>> 

We are having trouble finding this module, have you guys seen this before?

Thanks,

Alex

cmichelen, 2016-03-07:
It looks like you have to setup a few environmental variables (http://www.simpleitk.org/Wiki/ParaView/Python_Scripting#ParaView_and_Python). 

I will look at installing ParaView from scratch and update the documentation today.

cmichelen, 2016-03-08:
I was not able to reproduce your error. I installed ParaView (version 5) from scratch on a Windows machine and it worked fine (no extra step required).

Let us know if the solution on the link above works for you.

cmichelen, 2016-03-14:
We found out that our Macros are not working properly on the latest version of ParaView (v5) on Macs. Until they fix whatever is wrong revert to version 4.4 on Macs.

closed 2016-03-14 by cmichelen