WEC-Sim/WEC-Sim #102: Interpolation issue on irregularImport

Sire79, 2016-03-16:
Hi,

I am still new on WEC-Sim, I am trying to run RM3 model with irregular imported wave data.
The issue I am having is that when I populate ndbcBuoyData.txt with the data from one of our buoyees I get the following error. I have tried to use same size vector as the one used in the example but no difference. Could anyone understand what is going on here and assist me?
"
Error using griddedInterpolant
The grid vectors are not strictly monotonic increasing.

Error in interp1 (line 161)
        F = griddedInterpolant(X,V,method);

Error in waveClass/irregWaveSpectrum (line 380)
                    S_f = interp1(freq_data,Sf_data,freq,'pchip',0);

Error in waveClass/waveSetup (line 102)
                    obj.irregWaveSpectrum(g)

Error in wecSim (line 105)
waves.waveSetup(body(1).hydroData.simulation_parameters.w,
body(1).hydroData.simulation_parameters.water_depth, simu.rampT, simu.dt, simu.maxIt, simu.g,
simu.endTime);
"

Regards
Siyanda

yuyihsiang, 2016-03-21:
Hi Siyanda

Have you run the case with the one in the example? Is it fine?

We have read other data in the past and they all went fine. Please send me your imported wave data. I am more than welcome to take a quick look.

Thanks,

Yi-Hsiang

Sire79, 2016-03-22:
Hi Yi-Hsiang,

Thank you for your response. I have ran the example and it was fine.
The problem is when I populate the input text file with my own data. I have attached ndbcBuoyData file with my data in it. I have also attached the original data from our buoy measurements.

Thanks, 
Siyanda
[Slangkop_wav.txt](https://github.com/WEC-Sim/WEC-Sim/files/183980/Slangkop_wav.txt)

[ndbcBuoyData.txt](https://github.com/WEC-Sim/WEC-Sim/files/183972/ndbcBuoyData.txt)

yuyihsiang, 2016-03-23:
Hi Siyanda

I took a look at your data, and based on the error message, it seems to me that you data need to be sorted so that the data is monotonic increasing in frequency. 

Thanks,

Yi-Hsiang

Sire79, 2016-03-24:
Hi Yi-Hsiang

Thank you for checking my data and responding.
I will sort it accordingly, but how will that affect the results?

Regards,
Siyanda

yuyihsiang, 2016-04-11:
By default WEC-Sim uses 1000 wave frequency to generate the irregular wave. When the irregular spectrum data is input, WEC-Sim assumes spectrum from the irregular wave data import has to be monotonic increasing. If it is not, the interpolation will go wrong and give the error message.

Thanks,

--Yi-Hsiang

closed 2016-04-11 by yuyihsiang