WEC-Sim/WEC-Sim #104: Compiling MoorDyn and running numerical simulations with WecSim on LINUX

GianWa, 2016-03-17:
Hi,
I am having problems compiling MoorDyn on Linux Ubuntu and the wecSim code, for what concerns mooring coupling calculations with MoorDyn, seems to not be compatible with Linux operative systems.
Did anyone get in to the same issue? 
Thus, I would like to kindly request the developers to implement this feature.  
Thank you

yuyihsiang, 2016-03-17:
We have noticed that. It seems to be a compiling issue. We will look into this.

--Yi-Hsiang

yuyihsiang, 2016-03-21:
A MoorDyn libaray file for Linux (Lines.so) was created and included in the mooDyn folder under ./source/functions

mattEhall, 2016-03-22:
Hi, I know at least one MoorDyn user has been compiling on Linux, and I incorporated the required adjustments into the source code.  But different Linux compilers may require other adjustments.  (The current suspect is an ad-hoc isnan function in this case.)  Ultimately this is a MoorDyn rather than WEC-Sim issue, so lets make sure it gets reflected in the main MoorDyn source (via email or pull request on github.com/mattEhall/MoorDyn).  Thanks!

yuyihsiang, 2016-03-22:
Hi Matt

We were able to compile the Moordyn on a Linux machine by changing one line of the codes in Moordyn, which is related to how isnan function is defined. I have asked Gian to check it and see if the lines.so file works with WEC-Sim. If it works, I'll sent you what we did.

-Yi-Hsiang

mattEhall, 2016-03-22:
Thanks Yi-Hsiang.  My impression is some compilers have an isnan function already available and others don't (which is why I added it).  I'd love to find something that works without source changes in both cases.  The easiest cross-platform fix may just be to rename the isnan function in the code and always use that one.

closed 2016-04-11 by yuyihsiang