WEC-Sim/WEC-Sim #106: Nemoh-Bemio-Wecsim

erikhall, 2016-03-22:
Hi,

I am just getting started with Wec-sim and I have encountered some errors. The tutorials run just fine, but I am trying to import my own Nemoh-output, which does not seem to work. 

In my .slx file, I dont use constraints or PTO, only body and GRF. I get the error: 
"Domain error. To compute complex results from real x, use 'asin(complex(x))'."

What may the problem be?

Thanks,
Erik

yuyihsiang, 2016-03-23:
Hi Erik

You need to have at least a constraint or a PTO to properly link the body to the GRF. If you are doing a free floating simulation,  you will need to use the 6DOF constraint  (or 3DOF if you only want to consider surge, heave and pitch). Let me know if it solve the problem.

--Yi-Hsiang

erikhall, 2016-03-23:
Hi Yi,

Thank you for the reply. I forgot to mention that I also have MoorDyn in my .slx file, connected between GRF and my body. Do I still need to use PTO or constraint in that case?

Regards,
Erik

yuyihsiang, 2016-03-23:
Hi Erik

I believe you still need a PTO or a constraint. MoorDyn was coupled with WEC-Sim by applying forces calculated from MoorDyn to a non-hydrobody and connect that body to the WEC using a fixed joint. Let me know if it solves your problem is you add a 6DOF or 3DOF constraint. If not, we can further investigate the Nemoh part.

Thanks,

--Yi-Hsiang

erikhall, 2016-03-23:
Hi Yi,

Thank you, it worked with a 3DOF constraint, but not with the 6DOF. Any idea why that may be the case?

Regards,
Erik

saghy, 2016-04-06:
Hello,
I have done a free decay oscillation for the flap type wave energy converter.
Sicne Nemoh calculates the density of the object based on the Buoyancy Formula, the density could be different from the real density.
Secondly, Nemoh uses the center of buoyancy instead of center of rotation for the calculation. As a result, the moment of inertia and hydrostatic stiffness will be different.
Because in this case, the center of rotation is different from the center of buoyancy.
As far as I concerned, Wamit uses the same assumption.
Finally, in wecSim, I defined the regular wave with height equal to zero. And I defined the initial movement.
The point is that, I have to define the period, in which the wecSim read the corresponding hydrodynamics. (Am I right?)
I used the repetitive trial and errors with different periods, near the expected period of the flap.
I was wondering if there is any other way to do free decay analyses.
Thanks for time and help you will kindly provide

Saghy

cmichelen, 2016-04-11:
Hi Saghy,

You should run the 'noWave' or 'noWaveCIC' wave cases rather than the 'regular' wave case. Also the NEMOH mass/density should not matter, just make sure you specify the correct mass in the WEC-Sim input file. Let us know if this works for you. 

Thanks,

Carlos

yuyihsiang, 2016-04-11:
Hi Saghy

There was a bug in the 6-DOF constraint. I have fixed it and updated the WEC-Sim on Github. Please download the latest version if you need to use 6-DOF. Thanks for finding that bug, we really appreciate that. Let us know if works for you or not.

Cheers,

--Yi-Hsiang

yuyihsiang, 2016-04-11:
Hi Saghy

Follow up on what Carlos said, BEMIO post-processed hydrodynamic coefficients data by normalizing them  with density. The reason is because that is what WAMIT gives you.  WEC-Sim then dimenssionalized all the terms again during the simulation using the given the water density. By default water density is 1000 kg/m^3. You can overwrite that in the input file by specifying

simu.rho = <new density value>;
 
Let us know if works for you or not.

Cheers,

--Yi-Hsiang

erikhall, 2016-04-12:
Hi,

I use the 'regular' waveclass, with the same slx-model that yuiyihsiang suggested, i.e. with a GRF, a Moordyn, a 3DOF constraint and my hydrodynamic body, and that works fine. But when I switch to 'noWave' it does not work anymore. Instead I get the following errors: 
_______________________________________________________________________________
Error using wecSim (line 217)
* ['Wavetube/Global Reference Frame/Solver
Configuration']: Error evaluating equations
at time 48.65.  Stopping simulation.  There
may be a singularity in the solution.  If
not, try reducing the step size (either by
reducing the fixed step size or by
tightening the error tolerances)

Caused by:
    Error using wecSim (line 217)
    * ['Wavetube/Floating (3DOF)/CONSTRAINT
    (Planar Joint)']: 'Wavetube/Floating
    (3DOF)/CONSTRAINT (Planar Joint)' has a
    degenerate mass distribution on its base
    side.
_______________________________________________________________________________

Furthermore, I dont feel confident with the value I use in 'noWaveHydrodynamicCoeffT'. I use 400 based on the fact that my simu.endtime is 400.

Why does it not work? 

Thank you for your help,
Erik

saghy, 2016-04-12:
Hello Markus and Yi-Hsiang
Thanks a lot for the information you have kindly provided
1-	As far as I concerned, the mass (density) of an object has a major role in the calculation of hydrostatic stiffness and mass matrix. That will be used as an input for the simulation.

Also I have done some sensitivity analyses to be sure  if the effect could be negligible, which was not in my problems in hand. So, I did rewrite both the hydrostatic stiffness and mass matrix, before using bemio.
2- about cenetr of rotation, that is so important as well, i have changed them, too.

(references examples for mass matrix and hydrostatic stiffness:
Theory and application of ocean surface waves, chapter 8, mass and hydrostatic stiffness matrix)
Wamit theory manual, Matrix of hydrostatic and gravitational restoring coefficients)
3-	Yes, I have noticed, the scaling function bemio.io.nemoh in pyton code. And I did not scale the hydrodynamic inputs.
just as a reminding, in case you have not corrected yet.: i remembered,that the description provided for both true and false scale were the same(mistyping maybe), 

4-	noWave' or 'noWaveCIC, 
thanks for your suggestion, I used them and I make comparison with the regular wave with zero height, as I expected the results were the same
.................
** For Erik**
5- I think you should put the expected period of your device, ( trial and error procedure)  in the 'noWaveHydrodynamicCoeffT', Therefore wecSim could use the appropriate hydrodynamic coefficients for the simulation

kmruehl, 2016-04-12:
@saghy the output from the BEM run is parsed by BEMIO and scaled accordingly to the following:
http://wec-sim.github.io/WEC-Sim/theory.html#boundary-element-method
You should not need to modify your BEM output based on mass. Most BEM solvers calculate mass based on displaced volume, unless this default is overridden by the user. I recommend looking into this further to make sure the scaling is not being double accounted for. As @yuyihsiang mentioned, in WEC-Sim you can change the scaling by overwriting the default density using, simu.rho = ;

@erikhall waves.noWaveHydrodynamicCoeffT should be set to the period of response of the device. This need to be in the range of the wave periods you have BEM results for. The reason for your error is that 400s is likely out of the range of your BEM solution.

saghy, 2016-04-19:
Hello, 
Thanks for your response
I came across other problems, i will be grateful if you kindly advise me about them
1- when I do free decay oscillation, if I define the period (in weciSimInputFile) as a fist period it was defined for the BEM solver, wecsim considers it out of range.
Although in the **h5** file,i could see the coefficients for this period
Is there any limitation, when wecSim read wave information (upper and lower limit)?

2- since, i did a simplest possible model for free decay, without other source of damping, 
I have expected that the result will be the same, if i use the hydrodynamci coefficient and solve the following equation for one degree of freedom
(A+I)xtt+Cxt+kx=0
A=Addedmass
I=moment of inertia
k=hydrostatic stiffness
C=radiation damping
the following figure, shows the time series response.
As you could see, two time series have the same pattern, but the period is different. and i can not figure out why it happens.
from some approximate formula, it seems that the period must be approximately the same as the solution for  dynamic equation 
[image]

erikhall, 2016-04-20:
Hi,

Thank you for the reply. I have encountered other problems. I have re-performed the Nemoh-calculations, and this time they should be correct. But, apparently only the rotational constraint works for me in WEC-sim. When using any other constraint, the simulation stops. I get the following error:

"There may be a singularity in the solution.
If not, try reducing the step size (either
by reducing the fixed step size or by
tightening the error tolerances)"

The constraint location that I use is at center of gravity for the device. Should the constraint location not be at the center of gravity when using any of the constraints except the rotational? 

Or what may the problem be?

Thank you sincerely for you help.
Erik

saghy, 2016-04-20:
Hello Erik,
As I understood, you have defined a overdetermined system. I mean the number of the equations (constraints ,...) are more than the number of degree of freedom. 
In this case, wecSim could not solve the equation, there is no solution
So you got this error.

erikhall, 2016-04-21:
Hi saghy,

Thank you for the reply. I am not sure I understand though, my device is calculated for 6 degrees of freedom in Nemoh, so it should be able to cope with any constraint, not only rotational?

Regards,
Erik

erikhall, 2016-04-21:
Hello again,

I downloaded the latest release of the source code and now the 3dof-constraint works again. However, if I only want to constraint the device to heave and pitch, how might I do that?

Thanks,
Erik

saghy, 2016-04-21:
Hello Erik
when you are modeling in nemoh, you consider a floating body, without any constraints.....however when you use wecSim to mathematically reproduce the physic,  you should care about the degree of freedom and the kinematic and static constraints, ...the body is considered rigid bodies, so you just have dof in nodes...if you define constraints more than is necessary, it will be overdetermined, and simply it could not be solved. I hope , I could make it clear....
However, it is my idea about what you have said,  and maybe it has another reason.

erikhall, 2016-04-21:
Hi Saghy,

Thank you for the reply. I understand your logic and will take it into consideration when performing simulations. I managed to perform a simulation now with translational constraint and rotational PTO, and now it works. 

Thank you.

kmruehl, 2016-04-27:
@erikhall @saghy  have your questions been answered in this issue? May I close it?

-Kelley

saghy, 2016-04-27:
Hello, no. not yet

saghy, 2016-04-27:
Hello, 
Thanks for your response
I came across other problems, i will be grateful if you kindly advise me about them
1- when I do free decay oscillation, if I define the period (in weciSimInputFile) as a fist period, which was defined for the BEM solver, wecsim considers it out of range.
Although in the h5 file,I could see the coefficients for this period
Is there any limitation, when wecSim read wave information (upper and lower limit)?

2- since, I did a simplest possible model for free decay, without other source of damping, 
I have expected that the result will be the same, if i use the hydrodynamic coefficient and solve the following equation for one degree of freedom
(A+I)xtt+Cxt+kx=0
A=Addedmass
I=moment of inertia
k=hydrostatic stiffness
C=radiation damping
the following figure, shows the time series response.
As you could see, two time series have the same pattern, but the period is different. and I can not figure out why it happens.
from some approximate formula, it seems that the period must be approximately the same as the solution for dynamic equation 
[image]

erikhall, 2016-04-28:
I also have one final question, perhaps a bit far-fetched since it is about Bemio. I use Bemio to convert between formats (from GDF to dat for use in Nemoh). There is a python script in the Bemio-tutorial for this process. But that python script only considers half of the body, no symmetry. How would a python script be written to create a symmetrical body? (The second number in the .dat-file should be a 2, the python script writes a 1 by default)

My python script is written as follows:
----------------------------------------------------------------------------------
from bemio.mesh_utilities import mesh
import numpy as np

# Read WAMIT mesh
Geometry = mesh.read(file_name='body.GDF')

# Save to a NEMOH mesh
Geometry.write(mesh_format='NEMOH')
----------------------------------------------------------------------------------

Thank you for your help
Erik

erikhall, 2016-04-28:
I apologize for the headlines, I have no clue how that happened

Regards,
Erik

kmruehl, 2016-04-28:
@lawsonro3 or  @cmichelen can you respond to @erikhall's question about bemio?

kmruehl, 2016-04-28:
@saghy 
1. The min period from your BEM solution should work. I recommend double checking the units and rounding to make sure you're using a value that is indeed in range and can be interpolated from the BEM data. 
2. The WEC-Sim solution will only match the analytical solution if you are are truly simulating the exact scenario. The values you choose for your WEC-Sim hydro coefficients need to match the analytical solution exactly. This may or may not be applicable,depending on your model, but if there are any joints in your WEC-Sim model, these are going to impact your results. I do not know enough about your WEC-Sim model to be able to identify all of the possible sources of the discrepancy, but since the decay rate matches, and the period doesn't, that tells me the difference has to do with your stiffness or mass terms since omega_n = sqrt(k/m).

erikhall, 2016-04-28:
Hello again,

Never mind my question, I found an easier solution to my problem by just re-writing the dat-file manually.

Thanks

kmruehl, 2016-04-28:
@erikhall okay, :)

erikhall, 2016-04-29:
Hi again,

I am looking at the heave response for wave height H=1 and wave period T=20. I am comparing this response with my RAO for heave from Nemoh, and I can see that while my RAO goes towards 1 for low frequencies (which is true), my heave response amplitude in WECsim is at ~0.9. Does anyone have a theory of why this is and how to fix it?

Thanks a lot,
Erik

yuyihsiang, 2016-05-03:
Hi Erik

Do you have PTO damping in your WEC-Sim simulation?

Thanks,

--Yi-Hsiang

erikhall, 2016-05-03:
Hi Yi,

I managed to solve my problem. Apparently, I had set the scale parameter to 'False' in Bemio, but it should have been set to just False (or True) without the ''. Those things make Bemio read it as a string, and therefore it does not scale it, yet it needs to be scaled, either with False or True in order for the motion response in WEC-sim to be accurate.   

Thanks,
Erik

kmruehl, 2016-08-16:
FYI, the pre-latest release of WEC-Sim, [V2.1-pre](https://github.com/WEC-Sim/WEC-Sim/releases/tag/v2.1.1pre) has moved all of the bemio preprocessing into matlab. The new functions are located [here](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions/BEMIO).

closed 2016-08-16 by kmruehl