WEC-Sim/WEC-Sim #108: PTO User Defined Force/Internal Mechanics Question

ahagmuller, 2016-04-21:
Hi,

We are modeling a point absorber starting with a translational Z constraint for simplicity. In this setup, is the internal spring of the PTO (pto.k) in series or parallel with the buoyant spring of the device? 

If we were to apply a spring force, derived from a spring constant in n/m multiplied by displacement of the PTO to the "user input force" of the PTO block, would this spring force then be in series or parallel with the buoyant spring of the device?

To confirm, is the user input force in newtons?

Thanks,

Alex

kmruehl, 2016-04-27:
I presume your question about series or parallel is in reference to how the forces are accounted for in WEC-Sim, whether you can make a series/parallel assumption, similar to in circuitry?

If you go under the mask the 'Rigid Body' block, you will see that all of the forces are summed together (like a series circuit). Similarly, if you go under the mask of the 'Translational PTO Actuation Force' block, you will see that all of the forces as summed like a series circuit.

-Kelley

ahagmuller, 2016-04-27:
Hi Kelley,
Thanks for the reply. Yes, I am referring to the circuit analogy.
When I looked at the block, I thought springs in series was what was happening. However, upon further review I believe this is not actually the case. My argument for it being springs in parallel is as follows:

Per the schematic below, for simplicity we are looking at a floating single body point absorber connected to a PTO which is connected to the ocean floor, constrained to operate in 1 DOF, heave.
[image]

If we say that the "buoyant spring" of the device can be replaced with a mechanical spring fixed to the ocean floor, and fixed at the body of the device, the system looks like this:
[image]

This is contrary to springs in series, in that the PTO is connected to the device and to the ocean floor, and the buoyant spring is connected to the device and the ocean floor as well. If it were actually springs in series, then the PTO would be connected to the ocean floor, then to the buoyant spring, and the buoyant spring to the body of the point absorber, as per below:
[image]

So, if it is actually springs in parallel, the whole system would look like this:
[image]

Which would mean that any spring added to the PTO would contribute to the total spring of the system like this:
[image]
Instead of like this:
[image]

Meaning any additional spring constant added to the PTO actually increases the total stiffness and therefore increases the natural frequency of the system, if nothing else is changed.

If the springs were truly in series, any additional spring constant added to the PTO would REDUCE the total stiffness and reduce the natural frequency of the system, if nothing else is changed.

Can you confirm this to be the case?

Thanks,
Alex

kmruehl, 2016-04-28:
Alex, 

Okay, I understand your question. It's a pretty fundamental question about the SimMechanics application of forces between joints and bodies, and is unfortunately not entirely straightforward to answer. The visual representation is helpful to understand the question, but ultimately the difference between series/parallel comes down to the math, and how the forces are summed. As you showed for keq, a series circuit is simple addition, and a parallel circuit is not. What is WEC-Sim? A bit of both. Ultimately this comes down to implementation, and where/how loads are applied. The best way to determine this is to go under the mask of the body and pto blocks to understand where/how forced are applied. 

For example with damping, if you apply damping directly to the body via 'body(1).linearDamping', this damping is applied directly to the body, in the body block, and would be a series analogy, because this is included directly in the body force summation in the 'additional linear damping force calculation' subsystem

[image]

Alternatively, if you apply a damping through the pto block, via 'pto(1).c' (or as an external user-defined force), this would be not be a series analogy, because the pto block and the body block are not summed directly as above. In this case, 'pto(1).c' is applied directly at the joint within the PTO block, which would then be connected to the body block SimMechanics, as shown below:

[image]

If you use the PTO external actuated force, this is also not applied directly to the body, as shown in the RM3 PTO-Sim example for the coupling of the force-actuated pto block below.

[image]

I know this isn't a simple answer, but ultimately it comes down to not only the user's application of the code, and also the coupling between bodies and joints in SimMechanics.

I hope this helps,
-Kelley

cmichelen, 2016-05-03:
@ahagmuller  your description is correct. The springs are in parallel as long as the PTO is connected between the body and the the reference frame. 

-Carlos

ahagmuller, 2016-05-03:
Hi Kelley and Carlos,
Thanks for the reply on this. I realize this is quite fundamental but as Kelley pointed out, depending on where the damping or spring is applied could yield very different results. 

That being said, in coupling with MoorDyn or applying any mooring spring or damping in the example I described above to a single body point absorber, would it be correct o say that the mooring spring would act in series and not parallel?

Cheers!
Alex

cmichelen, 2016-05-03:
Hi Alex, 

This is what  I think: 

In the simple example you have above, if you consider heave only, you can combine the springs in parallel (hydrostatic spring and PTO spring) to get the equivalent stiffness for a single spring. You can then use that for natural period calculations. If you then add another linear spring for mooring (mooring matrix w/only K33 term used) whether it is in parallel or series with the other springs depends on how you created the simulink model. If the mooring matrix is also connected between the reference frame and the body then all 3 springs are in parallel. 

In this single-body, 1DOF system I think the only way to have springs in series would be to connect the blocks (PTO or mooring) in series in the simulink model.

If you start allowing more DOFS and use non-linear stiffness (MoorDyn) obtaining an equivalent linear stiffness value might not be as easy as combining springs in series and parallel.

kmruehl, 2016-05-05:
Alex,

In discussing this with Carlos, I think the easiest way to think about the series/parallel analogy is to asses which displacement the spring is subject to. If the springs observe the same displacement, they would be parallel, if they are do not, they are in series or some combination of series/parallel. I think it ultimately comes down to how/where the stiffness/damping is applied. 

-Kelley

closed 2016-05-23 by cmichelen