WEC-Sim/WEC-Sim #109: .stl Editor to use the NonLinear Hydrodynamic Modeling Feature

Bipass, 2016-04-25:
Hello,

I'd like to use the NonLinear Hydrodynamic Modeling on the geometry that I am working on. So, I would like a suggestion on which software to use to generate a correct .stl file with a good mesh starting from the cad output files (.igs gor example).

Thank you in advance.

Best regards,
Bipass

yuyihsiang, 2016-04-25:
Hi Bipass

We are using Rhino to generate the stl file when nonlinear hydrodynamics option is used. There should be some other options. The general rule of thumb is to have the mesh as close to equilateral triangle as possible.

Thanks,

--Yi-Hsiang

Bipass, 2016-04-25:
Hi!
Thank you for the very fast answer.
But, there is no opensource or free .stl editor that you know or would suggest?

Thank you again,
Bipass

mattEhall, 2016-04-25:
Hi,  
  Wings3D (www.wings3d.com) is a straightforward 3D modeler that is open source and can export to .stl.  I've had luck with it (for other applications) in the past.
 - Matt

Bipass, 2016-04-25:
Thank you Matt! I'll try it

kmruehl, 2016-04-27:
Matt, Thanks for the STL file suggestion! We've been wanting to explore open source options for generating better STL files for nonlinear hydro, and haven't had the chance. at SNL/NREL we typically use Rhino (https://www.rhino3d.com/).
-Kelley

closed 2016-04-28 by kmruehl