WEC-Sim/WEC-Sim #110: WEC-Sim-2.0pre does not work?

wyg110sjtu, 2016-04-27:
Dear  WEC-Sim developers:

         I have downloaded and installed WEC-Sim-2.0pre and tried to use it in the MATLAB 2014b enviroment. However, when I typed in the "wecSim" in the command prompt, I got the following error warnings:

Error using simulationClass/loadSimMechModel (line 81)
block_diagram does not have a parameter named 'SimMechanicsOpenEditorOnUpdate'

Error in wecSim (line 216)
simu.loadSimMechModel(simu.simMechanicsFile);

I do not know why. Can you please help me solve this problem?

Sincerely,

John,  Ph. D

kmruehl, 2016-04-27:
Unfortunately SimMechanics is not backwards compatible, and Matlab 2014b will not be able to open the WEC-Sim v2.0 library, because it was developed in 2015b. Please upgrade your Matlab to the following configuration:
http://wec-sim.github.io/WEC-Sim/getting_started.html

If the issue persists, let us know. Thanks,
-Kelley

kmruehl, 2016-04-27:
I would also highly recommend downloading the latest release of WEC-Sim v2.0 (not v2.0pre), or cloning the Git repository:
https://github.com/WEC-Sim/WEC-Sim/releases/tag/v2.0

kmruehl, 2016-05-05:
@wyg110sjtu did this resolve your issue?

wyg110sjtu, 2016-05-05:
Yes.

wyg110sjtu, 2016-05-21:
Hi, Kelley and Yi-Hsiang:

   In your paper “ Recent Additions in the Modeling Capabilities for the WEC-Sim-v1.1 Wave Energy Converter Design Tool” 

You mention that “Currently, WEC-Sim allows for the state-space realization of the hydrodynamic radiation coefficients  

in the  time-domain (TD);”. However,  I have not  found the specific MATLAB code inside the WEC-Sim software for the 

time-domain (TD) state-space realization of the radiation forces. Can you please tell me in which sub-routine does 

the MATLAB code for the time-domain (TD) state-space realization of the radiation forces exist?  Have you utilized 

the MATLAB function imp2ss to perform the time-domain (TD) state-space realization of the radiation forces ?

     A related more general question is : Where (in which sub-routine of the WEC-Sim) is the MATLAB code for solving the WEC 6-dof equations of motion?

Sincerely yours

wyg110

 

https://github.com/WEC-Sim/WEC-Sim/releases/tag/v2.0

cmichelen, 2016-05-23:
State-Space is implemented in WEC-Sim Library using the Simulink State Space block. It is under:

``WECSim_Lib/Body Elements/Rigid Body/Hydrodynamic Body/Wave Radiation Forces Calculation/SS CI and Constant-Damping-CoeVariant Subsystem/State-Space Calculation ``

[image]

The coefficients (A,B,C,D) are calculated in BEMIO.

[image]

Also take a look at the function ``irfInfAddedMassAndDamping`` under the ``./source/bodyClass.m``.

In BEMIO the calculation is done in the file ``bemio/data_structures/bem.py`` in the function ``calc_ss_radiation`` under the ``HydrodynamicData`` class.

wyg110sjtu, 2016-05-24:
Many thanks.

 

 <https://cloud.githubusercontent.com/assets/5296181/15472512/c021a92c-20b8-11e6-9e8e-ec55e42975d1.png> screen shot 2016-05-23 at 7 32 06 am

 <https://cloud.githubusercontent.com/assets/5296181/15472519/c55da832-20b8-11e6-88be-b8dcee2363f6.png> screen shot 2016-05-23 at 7 32 16 am

closed 2016-08-16 by kmruehl