WEC-Sim/WEC-Sim #112: Aqwa to Bemio to WecSim

ThommyBee, 2016-05-20:
Dear wecSim / Bemio team,

I compared the same very simple model using results from Nemoh->Bemio->wecSim and Aqwa(release 17.0)->Bemio->wecSim.

Results from wecSim (Forces & Body Response/Motion) for monochromatic "regular" waves match perfectly when using either, Nemoh derived h5 file or Aqwa derived h5 file.

However, results for monochromatic "regularCIC" simulation are off by a factor of 100 ONLY when using the Aqwa derived h5 file. Results match the "regular" wave case when using the Nemoh derived h5 file.

Using state space representation in wecSim does not change anything.
Both, nemoh as well as Aqwa mesh is fine enough as solutions do not change with increasing mesh fineness any more. 

While running bemio I ensured that all parameters for calculating the irf.excitation, irf.radiation, and ss_radiation are the same for both cases. Bemio commands are used with scale=False.  All bemio codes run through without any errors. 

When I use python commands to plot the excitation or radiation IRF over time it differs significantly in the order of magnitude (see attachment)

e.g.
IRF Radiation from Nemoh calc
[image]

vs, IRF Radiation from AQWA calc
[image]

Has anyone else experienced issues with Aqwa to Bemio? 

Thanks
Tom

cmichelen, 2016-05-23:
Try pulling the latest version of BEMIO. A few weeks ago a similar issue was resolved: https://github.com/WEC-Sim/bemio/issues/37

I suspect that should also fix the problem you are seeing. But if it doesn't let us know and we will take another look.

ThommyBee, 2016-05-25:
Thanks! The bemio version was updated right before these test cases. 
However, I think the problem was created by choosing multiple additional frequencies to calculate the BEM solution inside of Aqwa next to the standard range. (Standard range is suggested by Aqwa based on the mesh resolution applied to the structure and for most cases this range is probably sufficient.)

Possibly that could resolve into wrong infinite frequency added mass as I read in the other post that for Aqwa the inf. freq added mass is derived from the maximum frequency. I will try to do more tests soon and let you know if this was really the issue.

@kmruehl  This topic can be closed for now.

Thank you very much!
Tommy

closed 2016-05-25 by cmichelen