WEC-Sim/WEC-Sim #113: time-domain state-space realization

wyg110sjtu, 2016-05-21:
Hi, Kelley and Yi-Hsiang:

      In your paper “ Recent Additions in the Modeling Capabilities for the WEC-Sim-v1.1 Wave Energy Converter Design Tool” You mention that “Currently, WEC-Sim allows for the state-space realization of the hydrodynamic radiation coefficients  in the  time-domain (TD);”. However,  I have not  found the specific MATLAB code inside the WEC-Sim software for the time-domain (TD) state-space realization of the radiation forces. Can you please tell me in which sub-routine does the MATLAB code for the time-domain (TD) state-space realization of the radiation forces exist?  Have you utilized the MATLAB function imp2ss to perform the time-domain (TD) state-space realization of the radiation forces ?
A related more general question is : Where (in which  sub-routine of the WEC-Sim) is the
 MATLAB code for solving the WEC 6-dof equations of motion?

Sincerely yours

wyg110

cmichelen, 2016-05-23:
The actual 6DOF solver is done by SimMechanics. We apply the forces to the different bodies at each time-step. In the WEC-Sim library see ``WECSim_Lib/Body Elements/Rigid Body/Hydrodynamic Body``:

[image]

And ``WECSim_Lib/Body Elements/Rigid Body/Hydrodynamic Body/Structure``:
[image]

wyg110sjtu, 2016-05-24:
Many thanks.

 

 <https://cloud.githubusercontent.com/assets/5296181/15473028/3124d87c-20bb-11e6-839f-0f479cc0a77b.png> screen shot 2016-05-23 at 7 51 39 am

 <https://cloud.githubusercontent.com/assets/5296181/15473044/3f41be0c-20bb-11e6-9e5c-873a005bd8d6.png> screen shot 2016-05-23 at 7 52 02 am

closed 2016-08-16 by kmruehl