WEC-Sim/WEC-Sim #114: Ways to implement NL hydrostatics

mmacn, 2016-06-02:
If the buoyant forces of a body are known as a function of position, can this be implemented in such a way that wecSim doesn't have to use the body's CAD file to compute nonlinear hydrostatics? For example using a lookup table?

Thanks very much for your assistance

yuyihsiang, 2016-06-02:
Hi

Technically, it is possible to do so. However, in order for anyone to create this nonlinear hydrostatic data, he/she will need the CAD geometry (with good quality of panels) to generate that. In addition, if the nonlinear hydrostatic is used, it is also recommended to calculate the corresponding F-K force based on the device instantaneous position and wave elevation unless there is no wave. Based on our experience, the error form the nonlinear model can be larger than the linear mode if the nonlinear terms are not accurate.

I assume you have your WAMIT/nemoh/AQWA mesh file. I would recommend take that and create a stl file for the nonlinear hydrodynamics calculations. Is there any particular reason that you need to use a lookup table? 

Cheers,

--Yi-Hsiang

kmruehl, 2016-08-31:
@mmacn does this answer your question?

closed 2016-11-15 by kmruehl