WEC-Sim/WEC-Sim #118: NL Hydro - Negative Hydrostatic Pressure?

bradling, 2016-08-08:
When running `simu.nlhydro = 2;`, it appears that some cells have a negative hydrostatic pressure. Below is a histogram of all hydrostatic pressure values for an irregular wave case. 

I am having a hard time understanding how there could be negative values for hydrostatic pressure. Is there a physical meaning to these negative values, or is it an artifact of either the wheeler stretching or another numerical cause?

`hist(data.output.bodies(1).cellPressures_hydrostatic(:),100)` yields

[image]

yuyihsiang, 2016-08-08:
Hi Brad

The hydrostatic term represents the static part of the pressure from the hydrodynamic interaction. It is just the -rho*g*z term. The value will be negative if z is larger than zero. The dynamic part the pressure from the waves is calculated in the nonlinear FK force when the body is subject to waves. This is why simu.nlhydro = 2 is suggested for nonlinear hydrodynamic calculations.

-Yi-Hsiang

bradling, 2016-08-08:
Ok I think I understand now. Thanks for the explanation.

closed 2016-08-08 by bradling