WEC-Sim/WEC-Sim #119: Multiple Condition Runs,

Sire79, 2016-08-11:
Hi WECSim team,
I am attempting to simulate a device with a multiple condition runs feature (wecSimMCR) with the wave data in an excel spreadsheet. 
My questions are as follows:
-Should the data in the spreadsheet be a matrix or vector (i.e. wave power matrix or wave heights and     wave time stamps)?
-How do I call the spreadsheet data to the model? 

Regards
Sire

kmruehl, 2016-08-16:
There are many ways to run wecSimMCR depending on your application.  In order to use Option 3, as outlined in the [documentation ](http://wec-sim.github.io/WEC-Sim/features.html#multiple-condition-runs), your input file needs a line such as:
`simu.mcrCaseFile = 'heaveDecayCase.mat';`

The referenced [*.mat file is uploaded here](https://github.com/WEC-Sim/WEC-Sim/tree/exp/experimental_runs/1_freeDecay/heaveDecay) so that you can see the appropriate data structure.

kmruehl, 2016-08-16:
@yuyihsiang can you address the how to apply Option 2?

Sire79, 2016-08-17:
Thank you @kmruehl, I will run those examples for option 3 while I await for Option 2 to be addressed by @yuyihsiang . 

Much appreciated!

ThommyBee, 2016-08-22:
@Sire79 In general, simulink models can also be called without any problems if you write the small matlab routine yourself and use the "wecSim" command inside of any kind of loops. 

If the application, for example, is rather optimization than deriving a power matrix, you might want to use smarter "optimization algorithms" of parameters instead of shier mass sweeping, which can get quite time consuming depending on the model.

e.g. for a "regular" sweep

`
while "Loopcondition"
        % Here Definition of wecSim Input File Parameters e.g. k and c
        wecSim
        pause(5)
        % Post Processing and Optimization Algorithm to chose next loop iteration parameters"
`

@yuyihsiang @kmruehl Do you have recommendations for setting up the wecSim environment in a parfor loop on multiple "clients"?

Thank you very much
Thomas

yuyihsiang, 2016-08-23:
Hi Sire79

Q1: Option 2- It should be a matrix with 1st row is Tp, 1st column is Hs and the others are the occurrence for each sea states . The purpose to have this option is for people generating power matrix. If the wave environment has 0 occurrence, WEC-Sim won't perform simulation for that sea state.

Q2: Yes, you can write your own. In fact that is how wecSimMCR is developed. The file is located inside the function folder if you are interested to see.

kmruehl, 2016-08-31:
@ThommyBee @Sire79 have your MCR questions been resolved?

closed 2016-10-25 by kmruehl